riscv/riscv-crypto #175: Question on vl=vtype.vsew/128

topperc, 2022-08-18:
Multiple instruction definitions use the sentence `In order to properly specify the number of elements, vl needs to be set to vl = vtype.vsew/128`. Isn't vtype.vsew always a value less than 128? So this division would result in a value less than 1. Is this equation correct?

nibrunie, 2022-08-19:
I was also surprised by this phrasing. 
We have other description with an element group count `eg` (e.g. https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vaesef.adoc)
which seems clearer 
```
eg = (vl/EGS)
```
vl must be a multiple of `EGS`, so vl must be set to `eg * EGS = eg * EGW / EEW`

kdockser, 2022-09-07:
The equation is _not_ correct. Unfortunately, the equation was mid-edit when it was blindly copied and pasted elsewhere.
This document is still a work in progress and some of the definitions are closer to the final form than others.
I am replacing this incorrect equation with some better and more consistent wording - which will probably be tweaked as the spec is further refined.

kdockser, 2022-09-07:
Cleaned up. New text is better but will likely be further refined.

closed 2022-09-07 by kdockser