riscv/riscv-crypto #176: Additional constraints on vl and vstart?

JamesKenneyImperas, 2022-08-19:
The Crypto Vector Element Groups section says this:

> As with all vector instructions, the number of elements processed by a vector instruction is specified by the vector length `vl`. The number of element groups operated upon is then `vl/EGS`. If this ratio is not an integral value, an illegal instruction exception is raised.
> 
> Since `vlstart` is expressed in elements, the starting element group is `vstart/EGS`. If this ratio is not an integral value, an illegal instruction exception is raised.

I think there are other constraints on `vl` and `vstart` that are not covered by this. For example, using `vsetvli t0, a0, e8, ta, ma` on a machine with `VLEN=128` could set `vl` to `16`. For an instruction with `EEW=32` and `EGS=4`, this satisfies the integral value constraints, but must I think be illegal because 16 elements would require four vector registers to hold them if `EEW=32`.

Are constraints of the maximum values of `vl` and `vstart` described anywhere? This is the first time I've looked at this specification so it's likely I've missed something: apologies if so.

Thanks.

nibrunie, 2022-08-19:
That is a good point. I think here this should be understood as: this is an extra condition which raises an illegal instruction exception and other restrictions still applies.
In particular, most instructions specify that EEW is independent for SEW and EEW is defined by the instruction (.e.g sha256mc defines EEW=32 while sha512mc defines EEW=64), so both vl and vtype.vlmul should be valid:
- `eg = vl / EGS ` should be an integral value
- `eg * EGW <= vtype.vlmul * VLEN` should hold
- `vtsart / EGS` should be an integral value

JamesKenneyImperas, 2022-08-20:
An alternative might be to raise an Invalid Instruction trap if the instruction is ever executed with SEW != EEW defined by the instruction, which feels simpler. Is there an implementation benefit to allowing execution with SEW != EEW like this? Perhaps @aswaterman or @kasanovic could give their expert view on the implications for the vector engine (I'm not a hardware engineer, and also don't know much about how these instructions might be used in practice).

Thanks!

nibrunie, 2022-08-20:
Not an expert view: allowing `SEW != EEW` in such cases (the EEW is forced by the instruction) means we can skip some `vset*` changes if mixed element widths are involved (we still need to rely on them for `vl` setting though).
There are already a few instructions forcing EEW:
- most load and stores, e.g. `vle8`
- some permute force index EEW: `vrgatherei16.vv`

You are right, experts' opinion will be interesting here

jnk0le, 2022-08-20:
Those EEW forcing instructions keep the EEW/ELMUL ratio the same as SEW/LMUL.

kdockser, 2022-10-30:
Thanks for pointing this out, James. Along the lines of what jnk0le pointed out, what is missing is the statement of how EMUL is calculated. I borrowed the following text from the Vector spec (**7.3. Vector Load/Store Width Encoding**) and placed it into the Vector Crypto spec (**Crypto Vector Element Groups**)

> The corresponding EMUL is calculated as EMUL = (EEW/SEW)*LMUL. If the EMUL would be out of range (EMUL>8 or EMUL<1/8), the instruction encoding is reserved.

closed 2022-10-30 by kdockser

JamesKenneyImperas, 2022-11-25:
I think this issue might have been closed prematurely. I'm looking at the pseudocode for `vaesdf.[vv,vs]` and I can see the assertions checking that `vstart` and `vl` have valid multiples, so that's good. However, the number of iterations of the loop is defined by:

`eg_len = (vl/EGS)`

This isn't right I think: the value also needs to be clamped to a maximum value as described by Nicolas's post on August 19th, otherwise if the current `SEW` is small (e.g. 8) then `vl` could exceed the maximum allowed for the `EEW` defined by the instruction.

The same applies to all other instructions with fixed `EEW`.

I tried to reopen this issue but can't find a way to do that for some reason :-(

JamesKenneyImperas, 2022-11-25:
Also, there probably should be some indication of what happens to tail elements when `SEW=64` for instructions with fixed `EEW=32`. I assume this uses the normal `vtype.vta` control, but it would be good to state that.

Thanks.

JamesKenneyImperas, 2022-11-25:
In fact I think #237 addresses this, and it seems that that says that the policy for vl does match vector loads and stores in the base vector extension. Apologies.