riscv/riscv-crypto #179: Need clarification for vclmulh

EricCheng2222, 2022-08-29:
The result of vclmulh will be at most (2*SEW - 1) bits, does the "h" in instruction stands for high part(like high SEW bits)?
Also, if that's true, is it high SEW bits or high (SEW - 1) bits?

nibrunieAtSi5, 2022-09-01:
I think `h` stands indeed for the high part. This should be the `SEW - 1` highest bits, such that the concatenating of this result and the result of `vclmul` forms a the full (wide) result without overlap. 
The SEW-th bit will always be 0.

nibrunieAtSi5, 2022-09-01:
> thanks

You may want to wait for a more senior member to validate my answer @EricCheng2222

kdockser, 2022-09-07:
We will continue to update the definition of vclmulh. In the meantime, I believe that Nicolas' pull request addressed the question.

closed 2022-09-07 by kdockser