riscv/riscv-crypto #180: Need elaboration for vclmulgf.vi

EricCheng2222, 2022-08-29:
The spec reads:
`It treats each EGW=128 element group of vs1 and vs2 as the input polynomials. In order to match the GHASH specification, the bits need to be reversed - this includes the bits in each byte.`

I am pretty new to crypto, I don't know what is a GHASH spec, please help me understand, thanks. 
Also, in the operation, I don't see a reverse bit operation so I am a bit confused.

kdockser, 2022-09-07:
Hi Eric,
Thanks for your input. I agree that the definition of the vclmulgf instruction is very light at the moment. I am in the process of adding more detail including some initial pseudo code as well as information on external references.
Cheers,
Ken

kdockser, 2022-10-28:
This instruction has been replaced with vgfmac. The text has been clarified as to what the instruction actually does and includes a link the NIST specification to help people get a better understanding of GHASH.
Also, we later will add in sample RISC-V code that uses this instruction to properly implement GHASH.

closed 2022-10-28 by kdockser