riscv/riscv-crypto #182: Suggestion for different operand order for vsha512ms.v

nibrunieAtSi5, 2022-09-01:
The operand `vs1` of `vsha512ms.v` contains non contiguous words of the message schedule: `{W[11], W[10], W[9],  W[4]}`.

I would like to suggest the consideration of a different order `{W[4], W[11], W[10], W[9]} ` which could simplify implementation by datapath smaller than 256-bit wide, by requiring less operands when the instruction is cracked into more uops.

With this new split, the operation can be split into:
```
 W[16] = sig1(W[14]) + W[9]  + sig0(W[1]) + W[0];
 W[17] = sig1(W[15]) + W[10] + sig0(W[2]) + W[1];
```
and
```
 W[18] = sig1(W[16])   W[11] + sig0(W[3]) + W[2];
 W[19] = sig1(W[17]) + W[12] + sig0(W[4]) + W[3];
```
- the first half uses the high part of `vs2` the full `vd` (a), and the low part of `vs1`
- the second half uses the results of the first half, the low part of `vs2`, the high part of `vd` and the high part of `vs1`

With the orginal layout for `vs1`, both half needs the full `vs1`.

The drawback is that to build this new `vs1` layout a `vmerge` is no longer sufficient and an `LMUL=2` vslide1down may be required.

(a) except `W[3]`

kdockser, 2022-10-30:
Thanks for the suggestion, Nicolas.
My concern is the performance cost of this change, especially for datapaths that are at least 256-bits wide. The `vmerge` is a simple instruction, whereas the `vslide1down` with `LMUL=2` is more complicated instruction that will typically cost many more cycles to execute. This is already reason enough to stay with the current approach.
There is further penalty when `LMUL` is not already 2 (which will be common for larger implementations), as there will be the cost of `vsetvl`-type instructions to change `LMUL` to 2 for the `vslide1down` instruction and then back to the previous LMUL.
Therefore, my strong preference is to stay with the current approach.

nibrunieAtSi5, 2022-10-31:
@kdockser I agree with your arguments. It does not make much sense to slightly speed up a specific case and slow down any other.
closing this as "Wont' do"

closed 2022-10-31 by nibrunieAtSi5