riscv/riscv-crypto #186: round index in SM3C

nibrunieAtSi5, 2022-09-13:
The current version of `vsm3c` specification (https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vsm3c.adoc) states

> The immediate field rnd has only two valid values: 0 and 1; all other values are reserved. A 'rnd' value of 0 is used for rounds 0-15, and 1 is used for rounds 16-63.

Although this is enough to select the `T_j` round constant, it seems a 5-bit round index value (`j % 32`) is still required to perform the correct `T_j` rotation `(T_j <<< (j mod 32)))`. It also seems that the 5 bits may not be enough, since they are 48 possibles values for `T_j <<< (j mod 32)` (16 values for `0 <= j <= 15` and 32 values for `16 <= j <= 63`).

Initially reported to me by @ozturkerdinc 

References:
- https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vsm3c.adoc
- https://datatracker.ietf.org/doc/html/draft-sca-cfrg-sm3-02

nibrunie, 2022-09-14:
pointed out by @ozturkerdinc, as Instruction is computing two rounds, providing half the first round index as immediate is enough (and fits on 5 bits).
'rnd' 0 to 7 corresponds to first constant and round 0 to 15, 'rnd' 8 to 31 corresponds to second constant, round 16 to 63

nibrunieAtSi5, 2022-09-14:
The issue was also reported differently in https://github.com/riscv/riscv-crypto/issues/184

nibrunieAtSi5, 2022-09-15:
Update from Sept 15th TG meeting: @kdockser has fixed the spec to use half the round index as 5-bit immediate, closing this now.
Thank you @kdockser

closed 2022-09-15 by nibrunieAtSi5