riscv/riscv-crypto #190: The vsha2c output need to be the same in latest crypto draft 20220916

timhsu404, 2022-09-19:
The output format is dirrerent in description and pseudocode.
Which one should we follow?
```
vd = {a, b, e, f}; //in description
or
vd = {f,e,b,a); //in pseudocode
```

timhsu404, 2022-09-19:
@nibrunieAtSi5 EricCheng2222 for issue tracing.

nibrunieAtSi5, 2022-10-13:
@kdockser  it seems our ordering is not consistent in https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vsha2c.adoc

Should `a` be the least significant element (element 0, bytes 0 to 3 for SHA-256 and bytes 0 to 7 for SHA-512) or the other way around ?
How do we express this, I would have assumed: `{lowest element, ... , high element}` , is this correct ?

kdockser, 2022-10-18:
We are in the process of cleaning up the pseudo code. This is one of the items will be resolved in these changes.

kdockser, 2022-10-29:
This has been updated to be consistent between the pseudo-code and the specification.

closed 2022-10-29 by kdockser