riscv/riscv-crypto #192: Discussing vclmulgf alternatives

nibrunie, 2022-10-04:
@kdockser and @mjosaarinen  to continue the discussion on `GHASH`:

[image]

original diagram from wikipedia 

Assuming `Auth Data`, `Ciphertext <i>`, `H` as inputs and `Auth Tag` as output:
- the `MAC` is the best variant as it exactly tile the algorithm.
- the `ACCMUL` variant requires an extra initial `mult_H` and an extra final `xor` (the former could be implemented as a `vghash` with a `0`-value ciphertext

nibrunieAtSi5, 2022-10-06:
The GHASH spec  https://nvlpubs.nist.gov/nistpubs/Legacy/SP/nistspecialpublication800-38d.pdf specifies (page 12) the main operation as 
```
For i = 1, ..., m, let Yi = (Yi-1 ⊕ Xi) • H
```
This correspond to the ACCMUL variant, but hide the extra initial `mult_H` and final `xor`.

nibrunieAtSi5, 2022-10-06:
[image]

nibrunie, 2022-10-07:
Proposed variant in a multi-stream decomposition

== Multi-stream with MAC variant
```
H4V = {H^4, H^4, H^4, H^4}
HASH = {M_0, M_1, M_2, M_3}
for every 512 bits of message NM={M_4+i, M_4+i+1, M_4+i+2, M_4+i+3} // starting at i=1
   HASH = HASH  • H4V ⊕ NM
// eventually the signature must be reduce
RED = {H, 0, 0, 0}
// the following call to vclmulgf can be done on a single element group by limiting vl
HASH = HASH • RED ⊕ {M_4+i+1, M_4+i+2, M_4+i+3, 0} // addend is HASH slided down by 128 bits
HASH = HASH • RED ⊕ {M_4+i+2, M_4+i+3, 0, 0} 
HASH = HASH • RED ⊕ {M_4+i+3, 0, 0, 0}
HASH = HASH • RED ⊕ RED // final GHASH operation  
```

== Multi-stream with MULACC variant
```
H4V = {H^4, H^4, H^4, H^4}
HASH = {M_0, M_1, M_2, M_3}
HASH = (HASH ⊕ 0) • H4V  // initial operation with addend 0
for every 512 bits of message NM={M_4+i, M_4+i+1, M_4+i+2, M_4+i+3} // starting at i=1, excluding last chunk
   HASH = (HASH ⊕ NM) • H4V 
// last message chunk
NM={M_4+i, M_4+i+1, M_4+i+2, M_4+i+3}
HASH = (HASH ⊕ NM) • H4V 
// eventually the signature must be reduce
// this could be done with a 2 vredxor.vs e64, with proper masking (for odd/even 64-bit element) and than a vslideup
```

nibrunieAtSi5, 2022-10-13:
@mjosaarinen : the MAC version is better for implementation, as it allows the new message/ciphertext operand to be ready later than required in the ACCMUL variant.

nibrunieAtSi5, 2022-10-24:
We can close this discussion since the `MAC` version was selected by the task group.

closed 2022-10-24 by nibrunie