riscv/riscv-crypto #202: crypto-spec-scalar xperm4 xperm8 pseudo code fix

dansmathers, 2022-10-18:
both xperm4 and xperm8 pseudo code show:
foreach(i from 0 to xlen by

but it should be

foreach(i from 0 to (xlen-1) by

e.g. for xperm8 in rv32, 0, 8, 16, 24.   this is similar to what is currently checked in for the sail code.

kdockser, 2022-10-29:
This change should be made to the Scalar Crypto spec. However, I don't know how such clarifications  are to be made post-ratification. Does anyone know the policy on this?

cetola, 2023-02-21:
Updates to ratified specifications should be noted in an errata folder, and updated in a revised publication of the specification. See the [GitHub policy](https://docs.google.com/document/d/1RepGYxq-DBjHJYmy7fda9H3F9bB1MZiEbd-2m-ksijE/edit?usp=sharing) section containing "errata".

I'm going to work with Ken and Bill T to see that this gets updated.

Also, we will discuss this further on the [documentation SIG](https://lists.riscv.org/g/sig-documentation/message/57) if folks want to join in there.