riscv/riscv-crypto #206: Example RTL design

DuyTrandeLion, 2022-10-27:
Hello, do you have any reference example for rtl design? Or are you planning to share the example in the future?

kdockser, 2022-10-28:
We plan to provide sampled code to enable users to better understand how to use these instructions. We also plan to create SAIL code in the future after the Vector SAIL code has been approved and accepted. Some folks may choose to share their RTL code, but we do not have a plan to provide sample RTL code for the vector crypto extension.

closed 2022-10-28 by kdockser

DuyTrandeLion, 2022-11-20:
Hi @kdockser ,
I was able to put the RTL into my design and run some test with it using the C language. I noticed that currently there are no C examples on [AES](https://github.com/riscv/riscv-crypto/tree/master/benchmarks/aes/zscrypto_rv32) for the zscrypto although there are [C functions](https://github.com/riscv/riscv-crypto/blob/70311c0b606b6c931e53852ab8cef84f1bbf55fa/benchmarks/share/riscv-crypto-intrinsics.h#L76) calling the instructions. May I ask will there be any C examples on AES for zscrypto?