riscv/riscv-crypto #208: Starting discussion on some redundancies / peculiarity of the current spec (vector bitmanip)

nibrunieAtSi5, 2022-10-27:
Discussed with @aswaterman and @topperc

several redundancies and points which could be clarified
- `vbrev8.` https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vbrev8.adoc : SEW has not effect
   -  should we force SEW to a specific value (e.g. SEW=8) and leave the other configurations as reserved (would force switching to e8)
   -  extend the specification such that we do a bit-reverse in bytes + byte-reverse in element (combining `vrev8` and `vbrev8`:
      - this has the same behavior as current `vbrev8` when SEW=8-bit
      - this offers more use for `vbrev8` 
- some encodings seem redundant, for example we could limit `vrev8` to SEW >= 32-bit
  - `vrev8` when SEW=16 is equivalent to `vrol`/ `vror` with same SEW and imm=8
  - `vrev8` when SEW=8 is a NOP

aswaterman, 2022-10-27:
Note that type punning between different EEWs will cause performance hiccups on machines that optimize register layouts differently for different EEWs.  So, if we expect `vbrev8` to be popular for SEW>8, we should probably keep the current definition to avoid such type punning.  If we expect it to only be popular for SEW=8, then restricting it as @nibrunieAtSi5 mentioned would be OK.

If we did decide to restrict vrev8 to SEW >= 32 and vbrev8 to SEW=8, then I could see merging them into a single code point.  But that really hinges on how we expect `vbrev8` to be used.

kdockser, 2022-10-28:
I agree with Andrew.
Since `vbrev8` will typically be used when wider SEWs are already in effect, we should allow this instruction to be used with all SEWs.  This avoids unnecessary vsetvl instructions and enables straight-forward masking at the element level. It is similar to what we do with other bitmanip-type instructions whose function doesn't vary with SEW.

While some of the SEW values on vrev8 provide the same functionality as other instructions, I don't see any advantage in  restricting which SEW values are valid. I prefer to not add the complexity of exceptional cases when this can be avoided.

closed 2022-10-28 by kdockser