riscv/riscv-crypto #217: Need to clarify the vbrev8.v description for how SEW effect it's behavior

timhsu404, 2022-11-03:
> In some ways this is similar to how the bit logical instructions operate in that SEW
> does not affect the function.

The description above is ambiguous that it did not describe what "function" means.
Such as following case:
`0x21212121` with SEW=16, vstart=0 vl=1
after vbrev8.v execution, it will become `0x21212184`?
Or become `0x21218484`? 

> While this instruction supports the `SEW`, the behavior on each byte operated upon does not vary.

The description in v0.1.2 seems more accurate, maybe we can use it again.

timhsu404, 2022-11-03:
@nibrunie @EricCheng2222 
Tag them for tracking the issue.

kdockser, 2022-11-04:
Hi Tim, thanks for point this out.

Just to be clear, the comment to which you refer is in a box and thus non-normative; it has no effect on the meaning on the instruction. It was just added to point something out that readers might find interesting or were wondering about. However, since it has caused confusion it should be changed.

closed 2022-11-04 by kdockser