riscv/riscv-crypto #219: Suggestion for rotation with immediate behavior change

nibrunieAtSi5, 2022-11-04:
### Context
Currently RISV-V vector crypto extensions defines two rotation instructions `vror` and `vrol` , SEW-wide right and left element-wise rotations. Each admits 3 variants `.vv` , `.vx` , `.vi` with the straighfoward specification.
https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vrol.adoc
https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vror.adoc

### Issue:
The `.vi`  uses the default 5-bit immediate as an unsigned rotation amount.
the imm=0 value is useless since it corresponds to the identity
for SEW = 64 bits,  rotation by 32 bits (left or right) is inaccessible.
It was suggested to shift the immediate by 1 such that accessible rotation becomes [1, 32] but this implies incrementing the rotation amount from the default value.

### Suggestion:
What if we remove `vrol.vi` (a)  and define a vector right rotation by `SEW/2 + imm` instead, let's call it `vroro.vi`   (right rotation with offset). The offset by SEW / 2 is easy to perform (this rotation amount wraps around, at most a one bit addition is required for any SEW) and it becomes possible to cover rotation by 32 for SEW=64.
`vrol.vi vd, vs2, imm`  can be implemented by `vroro.vi vd, vs2, SEW / 2 - imm[log2(SEW)-1:0]`  when `imm[log2(SEW)-1:0] <= SEW/2`  or by  `vror.vi vd, vs2, SEW - imm[log2(SEW)-1:0]`  when `imm[log2(SEW)-1:0] > SEW/2`:
For example
- `vrol.vi vd, vs2, 5` for `SEW=16` is implemented by `vroro.vi vd, vs2, 3` (which is a right rotation by 11 bits)
- `vrol.vi vd, vs2, 9` for `SEW=16` is implemented by `vror.vi vd, vs2, 7` (which is a right rotation by 7 bits)

### Drawbacks
- non-uniform use of op encoding for vrol / vror
- may need to mux between `vs1` and `vs1 <<< (SEW / 2)`  before applying the actual rotation

(a) (this was suggested by @aswaterman  rather than removing `vror.vi` since scalar bitmanip only has `rori`)

DavidYuSi5, 2022-11-04:
I think the idea is good but it will increase the "decoding effort" when reading/writing the assembly code.
`vrol.vi vd, vs2, 5` for `SEW=16` is
equal to `vror.vi vd, vs2, 11` and is
equal to `vroro.vi vd, vs2, 3` with context `SEW=16`

The rotate length of  `vroro.vi vd, vs2, 3` will depend on `SEW`, and the original `vror.vi`/`vrol.vi` do not.

kdockser, 2022-11-04:
This is an interesting idea and would close gap of not being able to rotate a 64-bit value by 32.
A simpler alternative that could achieve the same end would be to remove rol.vi and define ror.vi to use one of the funct6 bits as uimm[5].

If we kept the current encodings we could use instruction[27] as uimm[5]. Alternatively, we could also change the proposed encoding for ror.vi tp be funct6[5:0]=01010x where x would be used as uimm[5].

jnk0le, 2022-11-04:
Instead of ofsetting, the  imm=0 can be special cased to be 32 or SEW/2. Though attempt to do it in vsetivli failed for some reason.

Also, immediate rotation by 32 can be done by `vrev32`

aswaterman, 2022-11-04:
I think Ken's and Nicolas' suggestions result in identical hardware and are only descriptively different: Nicolas describes two instructions distinguished by a single funct6 bit, and Ken describes a single instruction with a 6-bit immediate, of which bit 5 happens to be in the funct6 field.  Ken's proposal is cleaner for assembly programmers.

I also thought about @jnk0le's approach, but (assuming under that regime we'd keep both vrol.vi and vror.vi) it doesn't save any opcode space versus the other proposals, and seems to me to be less clean than the others.

nibrunieAtSi5, 2022-11-08:
Discussion continues on the mail group: https://lists.riscv.org/g/tech-crypto-ext/topic/removal_of_vrol_vi_and/94897241

kdockser, 2022-11-09:
The specification has been updated by removing vrol.vi and modifying vror.vi to have a 6 bit unsigned immediate composed from instruction encoding bits 26,19:15

closed 2022-11-09 by kdockser