riscv/riscv-crypto #226: Tail policy behavior on vector crypto instructions

eopXD, 2022-11-15:
I believe vector crypto instructions has corresponding tail policy behaviors like the vector instructions [0]. I think this worths a side note under the Introduction section about this (maybe somewhere next to the note on "Crypto Vector Element Groups" [1]).

[0] [v-spec: 5.4. Prestart, Active, Inactive, Body, and Tail Element Definitions](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#54-prestart-active-inactive-body-and-tail-element-definitions)
[1] https://github.com/riscv/riscv-crypto/blob/master/doc/vector/riscv-crypto-vector-element-groups.adoc

eopXD, 2022-11-15:
I doubt that there would be a use case for tail undisturbed...

The vector instructions that are always unmasked (which `vm=0` is always reserved) in v-spec has been designed to be always tail agnostic regardless of the policy specified in `vtype.vta`, e.g. [0], [1], and [2].

Always-unmasked instructions in v-spec has `vd` that is a vector mask register, which is different from the case here. However following the same logic I think the instructions that are always unmasked in vector crypto can follow the case.

[0] [v-spec: 13.13. Vector Floating-Point Compare Instructions](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#1313-vector-floating-point-compare-instructions)
[1] [v-spec: 15.1. Vector Mask-Register Logical Instructions](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#151-vector-mask-register-logical-instructions)
[2] [v-spec: 15.4. vmsbf.m set-before-first mask bit](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#154-vmsbfm-set-before-first-mask-bit)

------

Anyway, just raising the issue to have it discussed and clarified in the spec :)

kdockser, 2022-11-17:
Thanks for raising this.
I agree that we want to clarify this in the spec. However I am wondering if there is any value in forcing these instructions to always be tail agnostic when other non-mask producing instructions don't have such a requirement.

Does anyone see an _advantage_ to making the cryptography-specific instructions (i.e., not Zvkb) tail agnostic?

nibrunieAtSi5, 2022-11-17:
I think forcing tail agnostic for other instructions was done mostly  to simplify  instruction producing mask with a bit granularity `vl` (since each result element is 1-bit wide). Not sure the same argument applies for vector crypto with SEW/EGW-wide element result.

kdockser, 2022-11-25:
The spec now states that these instructions follow the tail policy in effect, just like most other vector instructions.

closed 2022-11-25 by kdockser