riscv/riscv-crypto #237: Incompatibility of vl and LMUL values

nibrunieAtSi5, 2022-11-19:
In the specification, multiple times is mentionned

> This instruction ignores vtype.vsew.

e.g. https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vaesdf.adoc

What about the case where `vl` is not compatible with the set `LMUL` value given the instruction `EEW` ?
- should this raise an illegal instruction exception ?
- should this set `vill` ?
- should it be implementation defined ?

e.g.
```
vsetvli x2, 16, e8, m1  // <= this will set LMUL=1, vl=16
vaesfd.vv v4, v8        // this will disregard SEW=8 and consider EEW=32, but since vl=16 LMUL is not wide enough

```

nibrunieAtSi5, 2022-11-19:
I am going to answer my own question but this is defined in the "element group" section of the spec it seems:

 > The corresponding EMUL is calculated as EMUL = (EEW/SEW)*LMUL. If the EMUL would be out of range (
EMUL>8 or EMUL<1/8), the instruction encoding is reserved

kdockser, 2022-11-25:
It seems like there are more and more reasons to constrain the instructions with a mandated EEW to only be used with SEW=EEW.

Does anyone have a good reason why we should ignore SEW? If not, I am inclined to require these instructions only be used with SEW set to the EEW.

grnewell, 2022-11-25:
> EXTERNAL EMAIL: Do not click links or open attachments unless you know the content is safe

Hi Ken,

I think the original reason to ignore SEW was that changing it to a required fixed value (assuming it was not already the mandated value), and then changing it back afterwards seemed redundant and a waste of instruction space and perhaps also time.

The merits (or not) of this argument should show up in real-world code.  Are other vector operations happening that require a different SEW while the crypto-specific instructions are also being used?  For example, does  calculating anything outside the core crypto instructions when implementing an encryption mode (like CTR or ECB mode) use a different value of SEW?  Or, for doing the GHASH (or SHA2) authentication calculations that would naturally be interleaved with the AES instructions?

Could the instructions with required SEW values be defined to actually set and restore the CSRs in addition to their core function?  This might alleviate some of the difficulties you are encountering with having them remain as they were while the instruction is running.

Rich

G. Richard Newell
Assoc. Technical Fellow, FPGA Business Unit, Microchip Technology
(408) 643-6146 (office), (408) 882-4785 (mobile), +1 (925) 478-7258 (Skype, not monitored)
PGP: (2009 DSA-1024, ELG-4096) B751 FC13 8B4E 49DA 2270 35A2 20E4 E66A D0D0 2E34
     (2016 SSA-4096, RSA-4096) 65F5 CCD6 23B3 BD3D CEDE AB58 171F F4DE E7D0 3ECA

kdockser, 2022-11-26:
Thanks Rich.

In all of the code I have seen so far, the EEW of the instruction matches well with the SEW used by other non-crypto instructions that are used by the algorithm. The is true even for AES-GCM as a 32-bit counter is used, this is the same width as the EEW for AES. While someone could implement AES-CTR, for example, with a 64-bit counter, there would still need to be an intervening _vset{i}vl{i}_ instruction to change the _vl_ for the 64-bit elements as there would be half as many of them compared to the 32-bit elements.

Come to think of it, there will likely be an SEW (and vl) change when coding AES-GCM with the _clmul_ instructions as it would be more efficient to do 64-bit carry-less multiplies than 32-bit. However, changing SEW on the fly will not be that uncommon in vector code and will be relatively performant when using the _vset{i}vli_ form of the instruction.

It seems that the complexities of having the vector crypto instructions ignoring SEW is great and the saving are probably insignificant, especially since there will still need to be _vset{i}vl{i}_ instructions interspersed to change the _vl_.

nibrunieAtSi5, 2022-11-26:
As far as I can tell, requesting SEW to be equal to EEW for those instructions would mostly translate into making sure that SEW independent instructions (think load and store before and after the crypto sequence) are configured using the same SEW as the crypto instruction when defining `vl`
for example rather than doing
```
vsetivli x1, 16, e8, m1, ta, ma
vle8 v17, (plaintest addr)
vle8 v22, (key addr)
(... aes sequence ...)
vse8 v22, (ciphertext addr)
```
one may have to do 
```
vsetivli x1, 4, e32, m1, ta, ma
vle32 v17, (plaintest addr)
vle32 v22, (key addr)
(... aes sequence ...)
vse32 v22, (ciphertext addr)
```
so matching load/store EEW with the EEW/SEW expected by the crypto sequence (in that case 32-bit for AES)

@kdockser  If we want to enforce a particular SEW for AES for example, I think we need to be careful:
- We should have the same rule for `vghmac.vv` which currently ignores `SEW`
- We need to evaluate the use of `vclmul*` and `vaes*` instructions in the same sequence as the former only work with `SEW=64`, so we might fall into one of the case described by @grnewell where SEW may need to be switched more often because of the new constrain

a HMAC SHA-512 + AES would require SEW re-configuration (I don't know if this is even used or if the sequences might overlap)

> Could the instructions with required SEW values be defined to actually set and restore the CSRs in addition to their core function?  This might alleviate some of the difficulties you are encountering with having them remain as they were while the instruction is running.

Which CSRs are you thinking about @grnewell ? `vtype` ?

kdockser, 2022-11-26:
While it would be nice to be able to leverage the same `vsetivli x1, 16, e8, m1, ta, ma` for loading bytes as performing the AES rounds, we are stuck with the problem of the _vl_ needing to be different for the AES instructions. If we are performing a single 128-bit wide AES [en|de]cryption, we would need _vl_=16 for the load of 16 bytes whereas the AES instruction would need  _vl_=4 for the four _EEW_=32 words.

I agree that we need to look at all of the instructions where we say the SEW is ignored. I think the best solution is to only define them for SEW=EEW.

It is best to avoid having any additional instructions be able to update the _vl_ and _vtype_ CSRs. The implementation complexity would explode if we started to allow these to be updated by more than _vset{i}vl{i}_ (_vl_ and _vtype_) and the fault-only-first vector load instruction variants (_vl_).

nibrunieAtSi5, 2022-11-29:
> I think that the load instructions with specific EEWs can avoid the use of additional vset{i}vli only when the same vl can be used for the load and the subsequent instructions. For example, the load EEW can be 2x SEW when the loads are feeding a narrowing instruction; both use the same vl.

I think the main aspect which prevents us from using `v[ls]e32` memory operations to keep `vl` in (that I can see):
```
vsetivli x1, 4, e32, m1, ta, ma
vle32 v17, (plaintest addr)
vle32 v22, (key addr)
(... aes sequence ...)
vse32 v22, (ciphertext addr)
```
is the fact that plaintext and ciphertext may not be 4-byte aligned and implementation are not allowed to support vector misaligned access so `vle32 v17, (plaintest addr)` may trigger a mis-aligned exception on some implementation.
Else the same `vl` could be used, isn't it ?

kdockser, 2022-11-29:
As I recall, vector does allow implementations to either support misaligned element access or raise a an address misaligned exception. And, yes, we could use a load with EEW=32 with an AES instruction. However, this is a separate issue.

The point that I was trying to make was other instructions with implicit EEW are able to work (without an intervening _vset{i}vl{i}_ ) with instructions with a different SEW if both can use the same _vl_. In the case of an AES instruction, we need _vl_ to be 4 times the number of 128-bit AES operations that are being performed. This would not work with an SEW=64 instruction (for example) as it would need _vl_ to be 2 times the number of 128-bit  AES operations. The instructions need different vector lengths and therefore cannot avoid the intervening _vset{i}vl{i}_; thus for these crypto instructions there is no advantage to using an SEW that is different from the EEW.

grnewell, 2022-11-29:
Hi Ken,

I understand your point that vl has to be 4x, the way it is currently defined for element groups as the total number of sub-elements.

Why isn’t vl defined, instead, to be the number of element groups?  A element group is really more like our vector element. Then, vl might not have to be changed as often as the number of vector elements being worked on in a kernel is likely to be the same everywhere (e.g., number of AES blocks = number of counter elements = number of GHASH multiplies).

To say that a 128-bit AES message block is 4 x 32-bit sub-elements is totally arbitrary and doesn’t have any meaning to the algorithm.  And, it forces vl to be 4x larger than is natural, forcing all the extra vsetvl instructions, as you have pointed out. (It’s the same problem if you say it’s 2 x 64-bits).  But, if 1) vl were invariant (as I am hypothesizing), and 2) AES ignored SEW and always worked on 128-bits, and 3) any intervening vector instruction (e.g., updating the counters) could use whichever SEW makes sense (e.g., SEW=32).  (BTW, the GHASH multiply would also ignore SEW and always use the defined 128-bit GF.)

G. Richard Newell
Assoc. Technical Fellow, FPGA Business Unit, Microchip Technology
(408) 643-6146 (office), (408) 882-4785 (mobile), +1 (925) 478-7258 (Skype, not monitored)
PGP: (2009 DSA-1024, ELG-4096) B751 FC13 8B4E 49DA 2270 35A2 20E4 E66A D0D0 2E34
     (2016 SSA-4096, RSA-4096) 65F5 CCD6 23B3 BD3D CEDE AB58 171F F4DE E7D0 3ECA

nibrunieAtSi5, 2022-11-30:
> Why isn’t vl defined, instead, to be the number of element groups?  A element group is really more like our vector element. Then, vl might not have to be changed as often as the number of vector [...]

I think this may case newer problems: most implementation have not been made to support SEW > 64, if we say vl is the number of element groups, then it is equivalent to requesting implementation to support SEW=128 or SEW=256 (it terms of determining head / body / tail ...).

> To say that a 128-bit AES message block is 4 x 32-bit sub-elements is totally arbitrary and doesn’t have any meaning to the algorithm. 

I agree with that, it would make more sense to have **(E/S)EW=8* for AES (counting the number of bytes has an actual meaning).
For `SM3`/`SHA-256`/`SHA-512` (always the last two do not ignore `SEW`, I find it understandable to have `SEW` be assigned to the word size used by the algorithm

grnewell, 2022-11-30:
Hi Nicolas,

I wasn’t meaning to suggest that SEW actually be set to anything >64, just that AES would have an implicit EEW of 128 (regardless of the current SEW setting).

I agree that SEW of 8 (i.e., a byte) makes more sense for AES which internally has 8 bit “words”  than 32 (and using the “correct word size” of 32 or 64 for the hashes), but it’s still rather arbitrary as the algorithm is going to ignore the setting, anyway.   Then, if SEW=8, with the current definition of vl it would have to be 16x for AES, which doesn’t solve the problem of requiring frequent vl changes inside the loops.

My question is more towards the definition of vl.  If element groups are used, why shouldn’t vl be the number of element groups, rather than the number of subelements?  There may be a good reason, but I am not currently seeing it.

Rich

G. Richard Newell
Assoc. Technical Fellow, FPGA Business Unit, Microchip Technology
(408) 643-6146 (office), (408) 882-4785 (mobile), +1 (925) 478-7258 (Skype, not monitored)
PGP: (2009 DSA-1024, ELG-4096) B751 FC13 8B4E 49DA 2270 35A2 20E4 E66A D0D0 2E34
     (2016 SSA-4096, RSA-4096) 65F5 CCD6 23B3 BD3D CEDE AB58 171F F4DE E7D0 3ECA

nibrunieAtSi5, 2022-12-01:
Following the task group discussion of Dec 1st 2022, I am closing this issue.
- the TG has decided to remove "ignore SEW" from all crypto instructions which had it and switch to forcing the setting of a proper SEW value (to match the old `EEW` value) and consider every other `SEW` as reserved.

closed 2022-12-01 by nibrunieAtSi5