riscv/riscv-crypto #239: Vector cryptographic instruction tests

JamesKenneyImperas, 2022-11-24:
Are there any tests for the vector cryptographic instructions in this repository or elsewhere? I've floundered around for a while but can't see anything like this.

Thanks.

kdockser, 2022-11-25:
Are you asking about ACTs or code that implements the algorithms (e.g, AES-128) using the instructions and then compares the results to known answers? We are working on making the latter publicly available in the next week or two.

JamesKenneyImperas, 2022-11-26:
Those tests sound exactly what I want. Thanks.

Perhaps when the tests are ready you could close this issue with a pointer to them?

kdockser, 2022-12-01:
While I was hoping that we would have some basic ACT-like tests soon, the lack of vector support in the ACT infrastructure has been a major impediment that we have not been able to work around. My estimation of when these tests will be available was overly optimistic. At this point I do not have an ETA of when they will be available.

allenjbaum, 2022-12-02:
You can ask questions at the 4pm PT special arch-test SIG session
next Wednesday.
I'll send you the invite. I'll send it to the group, actually.

allenjbaum, 2022-12-02:
Special ACT meeting: Vector spec testing
Wednesday Dec 7, 2022 ⋅ 4pm – 5pm
Pacific Time - Los Angeles

Location
https://zoom.us/j/95412060804?pwd=REdXVFNtZElZUk9IUU9ESUM0QndFUT09	
https://www.google.com/url?q=https%3A%2F%2Fzoom.us%2Fj%2F95412060804%3Fpwd%3DREdXVFNtZElZUk9IUU9ESUM0QndFUT09&sa=D&ust=1670397480000000&usg=AOvVaw0WZMAPTaYlqKzAXxfSkM77

This is an off-cycle arch_test SIG meeting at a different time and day to  
accommodate China time-zones (sorry Europe).The meeting is being called to  
discuss vector spec tests. The preliminary agenda is:1. RVV ATG Status  
&amp; Coverage2. RISC-V ISAC extension for RVV3. PR submissionThe usual 2nd  
week arch_test SIG meeting will be held on the following day, Dec. 8 at the  
usual time (8am Pacific time)The agenda will be to continue reviewing open  
issues

Organizer
***@***.***
***@***.***

Guests
***@***.*** - organizer
***@***.***
***@***.***
***@***.***
***@***.***
Alexander Podoplelov
Li, Pan2
***@***.***
***@***.***
View all guest info  
https://calendar.google.com/calendar/event?action=VIEW&eid=NGtqcmVwcGFmZDFlcXF2cTg0djkzMWEwMWggcmVwbHkrYWhweHZqc3g1ZG1oZ2hzcG0yZG4zNW9ic3pkM25ldmJuaGhmb3BhdmttQHJlcGx5LmdpdGh1Yi5jb20&tok=MjgjYWxsZW4uYmF1bUBlc3BlcmFudG90ZWNoLmNvbTUzZDk2NmY1YmE0ODAxYjU0MmY0ZDc2YmFiMTk2ZWI2MDg3NjE0MDY&ctz=America%2FLos_Angeles&hl=en&es=0

Reply for ***@***.***  
and view more details  
https://calendar.google.com/calendar/event?action=VIEW&eid=NGtqcmVwcGFmZDFlcXF2cTg0djkzMWEwMWggcmVwbHkrYWhweHZqc3g1ZG1oZ2hzcG0yZG4zNW9ic3pkM25ldmJuaGhmb3BhdmttQHJlcGx5LmdpdGh1Yi5jb20&tok=MjgjYWxsZW4uYmF1bUBlc3BlcmFudG90ZWNoLmNvbTUzZDk2NmY1YmE0ODAxYjU0MmY0ZDc2YmFiMTk2ZWI2MDg3NjE0MDY&ctz=America%2FLos_Angeles&hl=en&es=0
Your attendance is optional.

~~//~~
Invitation from Google Calendar: https://calendar.google.com/calendar/

You are receiving this email because you are an attendee on the event. To  
stop receiving future updates for this event, decline this event.

Forwarding this invitation could allow any recipient to send a response to  
the organizer, be added to the guest list, invite others regardless of  
their own invitation status, or modify your RSVP.

Learn more https://support.google.com/calendar/answer/37135#forwarding

kdockser, 2023-01-11:
Closing as there is a waiver for Vector Crypto ACT at this time.

closed 2023-01-11 by kdockser