riscv/riscv-crypto #244: Is there a need for a non bit reversed `vghmac.vv` ?

nibrunieAtSi5, 2022-11-30:
It seems like the AES-XTS standard is using a galois field multiplication with the same irreducible polynomial as GCM but without bit reversing every byte.

[image]

if my understanding is correct, it could benefit from a modified version of `vghmac.vv` without bit reversal as this will save two `vbrev8` instructions.

### Reference
- https://luca-giuzzi.unibs.it/corsi/Support/papers-cryptography/1619-2007-NIST-Submission.pdf

nibrunieAtSi5, 2022-12-01:
Task group discussion: the AES-XTS should not need a full GF multiplication: it is just a multiplication (shift) and reduction when required.

kdockser, 2022-12-01:
Thanks for bringing this up Nicolas. As you noted in your reply, the TG decided that this was not needed.

closed 2022-12-01 by kdockser