riscv/riscv-crypto #246: On consistency of mnemonics "vv"

eopXD, 2022-12-01:
I think this is a matter of personal taste, but `vaesdf.vv` takes only 2 registers as input doesn't match formats of other "vv" instructions, like general instructions in v-spec, or the vector bit manipulation instructions.

I believe the reason for it to be that "vv" is that both `vd` and `vs2` are served as input in the instruction, but I think there is an inconsistency here with other vector instructions having `vs2` and `vs1` when it has a "vv" mnemonics.

kdockser, 2022-12-01:
You are correct, this is different from other two operand .vv instructions. It is closer to three operand .vv instructions in that it uses vd as a source. The funct6 encodings of these instructions are grouped with other .vv instructions that read vd as a source.
Also, this approach is consistent with vector unary instructions in that it uses the vs1 field as a part of the instruction encoding. The result is these instructions take up much less opcode space than the would have if they had followed the approach of other .vv instructions.

closed 2022-12-01 by kdockser

eopXD, 2022-12-02:
Thank you for explaining.