riscv/riscv-crypto #261: Summarizing illegal/reserved cases

nibrunieAtSi5, 2023-01-02:
Following some discussions with colleagues, it seems a summary of illegal / reserved cases might be useful for spec review (and possibly later). Here is a personal attemp:

[image]

# `vl` and `vstart` constraints

Because most vector crypto instructions use element groups they inherit from the constraint on vl and vstart values, namely they must be multiple of the element group size EGS:
- for `vaes*`, `vsha2*`, `vghmac`, `vsm4*`: `vl` and `vstart` must be multiple of `4`
- for `vsm3*`, `vl` and `vstart` must be multiple of `8`

# `SEW` constraints

vector crypto instructions enforce constraint on SEW, namely
- for `vaes*`, `vghmac`, `vsm4*`, `vsm3*`: `SEW` must be equal to 32-bit, all other cases are _reserved_
- for `vsha2*`
  -  `SEW` must be equal to 32-bit if Zknha only is implemented
  - `SEW` must be equal to 32-bit or 64-bit if Zknhb is implemented
  - all other cases are _reserved_
- for `vclmul` and `vclmulh`:  `SEW` must be equal to 64-bit, all other cases are _reserved_

> NOTE: In an older version of the draft specification, SEW used to be ignored

# Source/Destination overlap constraints

- For `vaes*.vs` (and `.vs` variant only): the single register `vs2` cannot overlap with the register group `vd`, the overlap case is _reserved_
- For `vsm4r.vs`: (and `.vs` variant only): the single register vs2 cannot overlap with the register group vd, the overlap case is _reserved_
- For `vsha2c[hl].vv`  and `vsha2ms.vv`: the case where the register group `vd` overlaps with either the register group `vs1` or `vs2` is _reserved_
- For `vsm3me.vv`: the case where the register group `vd` overlaps with the register group `vs2` is _reserved_
- For `vsm3c.vi`: the case where the register group `vd` overlaps with the register group `vs2` is reserved

# Immediate legal range

Some vector crypto instructions use the rs1 bitfield to indicate a round index, legal values are limited to a range defined for each instruction:

- `vaeskf1`.vi: `uimm` immediate value, indicating new round key index, can take any value from `1` to `10`, out of range values signal an illegal instruction exception
- `vaeskf2.vi`: `uimm` immediate value, indicating new round key index, can take any value from `2` to `14`, out of range values signal an illegal instruction exception 
- `vsm4k.vi`: `uimm` immediate value, indicating new round key group index, can take any value from `0` to `7`, out of range values signal an illegal instruction exception 

- https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vsha2c.adoc 
- https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vsha2ms.adoc 
- https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vsm3me.adoc 
- https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vsm3c.adoc

*

nibrunieAtSi5, 2023-01-02:
Note: for the immediate valid range, there exists a small difference between `vaeskf*` and `vsm4k`. The formers list that an out-of-range immediate value signals an illegal exception, while the later lists that the "out of range values are reserved". I don't think we intend any formal difference between the two so maybe it would be good to unify the description:

- https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vaeskf1.adoc
- https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vaeskf2.adoc
- https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vsm4k.adoc

kdockser, 2023-01-11:
Since it is hard to review a changing spec, we are limiting changes to help expedite the AR review.
The following changes have been made to address  inconstancies and to make constraints easier to determine:

1. Immediate values that were out-of-range were handled inconsistently with some triggering exceptions and others mapped to in-range values. This has been changed so that all map to in-range values
2. A separate heading was added for instructions with reserved encodings (SEW and overlap). Under this heading all reserved encodings are listed.

nibrunieAtSi5, 2023-01-11:
@kdockser what is the rationale behind negating the valid MSB when the index imm falls out of range for `vaeskf[12]` ?
(I may not be able to attend tomorrow meeting so I am asking here)

kdockser, 2023-01-11:
The primary goal here is to eliminate data-based invalid exceptions, and instead to define the behavior for all inputs including those that are out of range. One way this is done in RISC-V is to ignore the upper bits of the immediate (e.g., for shifts). However. this is not sufficient when we have out-of-range values that fit in the significant immediate bits. Inverting the most significant bit in out-of-range values is an easiest way to map these values to be in-range where we already have defined behavior.

kdockser, 2023-01-11:
A section that summaries Vector Crypto Constraints has been added to the spec.

closed 2023-01-11 by kdockser