riscv/riscv-crypto #262: vsm4r pseudocode looks garbled

JamesKenneyImperas, 2023-01-03:
I think the pseudocode for vsm4r is garbled, specifically the "B" term expressions for 2nd and 3rd rounds. Should they be this:

second round:
B = (X2 ^ X3 ^ **Y0** ^ rk1)

third round:
B = (X3 ^ Y0 ^ **Y1** ^ rk2)

I'm not an expert, so I'm guessing here. Can you please check?

Thanks.

JamesKenneyImperas, 2023-01-04:
There also seem to be mistakes in the pseudocode for vsm3c:

1. register vd is not in the function parameter list, but used in the function body;
2. a variable "round" is used - should this be "rnds"?
3. a function "FF" is used - should this be "FF_j"?

Is there _validated_ code somewhere that I can refer to?

Thanks.

nibrunieAtSi5, 2023-01-05:
Rivos (cc @kdockser ) has published a public pre-version of their spike fork extended with vector-crypto support: https://github.com/rivosinc/riscv-isa-sim/tree/zvk-vector-crypto

kdockser, 2023-01-05:
Fixed these errors in vsm4r and vsm3c.

closed 2023-01-05 by kdockser

JamesKenneyImperas, 2023-01-06:
@kdockser, can you please provide an example command line that should be used to enable Zvk instructions in your spike fork? I'm getting an illegal instruction trap on every one.

Thanks.