riscv/riscv-crypto #264: vsm3me.vv decode conflict with SRLI32.u

JamesKenneyImperas, 2023-01-04:
The decode for instruction `vsm3me.vv` conflicts with `SRLI32.u` defined in the P extension. Are these two extensions intentionally incompatible?

There are no other decode conflicts, as far as I can see.

Thanks.

nibrunieAtSi5, 2023-01-05:
Yes, some conflicts are expected as it has been decided that the P opcode space could be reused by some vector extensions as they are not designed to be compatible: 
See this email exchange in one of RISC-V technical groups
https://lists.riscv.org/g/tech-vector-ext/topic/overlapping_p_and_v_opcodes/95837243?p=,,,20,0,0,0::recentpostdate/sticky,,,20,2,0,95837243,previd%3D1672845520402831657,nextid%3D1644409485258595148&previd=1672845520402831657&nextid=1644409485258595148

kdockser, 2023-01-05:
Echoing what Nicolas said, Vector extensions, including the Vector Crypto Extensions, will be using the P opcode space.
The packed-SIMD "P" specification is not yet finalized - so any encodings in that specification are preliminary and subject to change.
That said, there is no guarantee that vector extensions and packed SIMD extensions will not have collisions in the P opcode space.

closed 2023-01-05 by kdockser