riscv/riscv-crypto #265: inconsistent implementation of vsha2ms.vv in rivosinc

JamesKenneyImperas, 2023-01-06:
The implementation of vsha2ms.vv in the rivosinc Spike fork does not match the specification. The specification says:

```
    {W[3],  W[2],  W[1],  W[0]}  : bits(EGW) = get_velem(vd, SEW, i);
    {W[11], W[10], W[9],  W[4]}  : bits(EGW) = get_velem(vs1, SEW, i);
    {W[15], W[14], W[13], W[12]} : bits(EGW) = get_velem(vs2, SEW, i);

```
But in the Spike fork it is:

```
        // {w3, w2, w1, w0} <- vd
        EXTRACT_EGU32x4_WORDS_BE(vd, w3, w2, w1, w0);
        // {w11, w10, w9, w4} <- vs2
        EXTRACT_EGU32x4_WORDS_BE(vs2, w11, w10, w9, w4);
        // {w15, w14, w13, w12} <- vs1
        EXTRACT_EGU32x4_WORDS_BE(vs1, w15, w14, w13, w12);

```
Note that vs1 and vs2 are swapped. *Can you please confirm which is correct*?

Thanks.

nibrunieAtSi5, 2023-01-06:
Another similar issue was raised yesterday (`vaseskf2` round index range) in RIVOS spike implementation.
I would say the spec is correct here, but maybe @kdockser can comment on that.

kdockser, 2023-01-06:
Thanks for pointing this out, James. I changed the spec to match SPIKE as this is a minor (but important) change and is much easier to do in the spec.

Nicolas, thanks for reminding me of the vaeskf2 round index range. This is part of a bigger problem (which you have alluded to in other issues). Basically, we have been inconsistent on handling out-of-range values; we ignore bits that are out of range (as is commonly done in RISC-V) but have been triggering exceptions on out-of-range values within the needed bits. I discussed this with Andrew and he suggested that we follow the V extension. This means that we will not signal exceptions on out-of-range immediates, and provide well defined behavior for all out-of-range values (immediate and register-read).

closed 2023-01-06 by kdockser