riscv/riscv-crypto #267: inconsistent implementation of vaeskf2.vi in rivosinc

JamesKenneyImperas, 2023-01-06:
The implementation of `vaeskf2.vi` in the rivos.inc Spike branch does not match the specification. The specification says:

```
      let w[3] : bits(32) = if (rnd[0]==1) then
        RoundKeyB[3] XOR aes_subword_fwd(CurrentRoundKey[0])
      else
        RoundKeyB[3] XOR aes_subword_fwd(aes_rotword(CurrentRoundKey[0])) XOR aes_decode_rcon(rnd>>1);
```

The Spike implementation has this:

```
    const bool is_odd_round = (round & 0x1) != 0;
    if (is_odd_round) {
      temp = (temp >> 8) | (temp << 24);  // Rotate right by 8
    }
    temp = (((uint32_t)AES_ENC_SBOX[(temp >> 24) & 0xFF] << 24) |
            ((uint32_t)AES_ENC_SBOX[(temp >> 16) & 0xFF] << 16) |
            ((uint32_t)AES_ENC_SBOX[(temp >>  8) & 0xFF] <<  8) |
            ((uint32_t)AES_ENC_SBOX[(temp >>  0) & 0xFF] <<  0));

    if (is_odd_round) {
      const uint32_t rcon = kRoundConstants[round >> 1];
      temp = temp ^ rcon;
    }
```

This implements the two branches if the 'if' statement in the opposite sense. Can you please confirm whether the specification is correct?

Thanks.

kdockser, 2023-01-06:
Thanks for bringing this up James. In this case the SPIKE code is wrong and fix has already been requested. Basically, the SPIKE code was numbering the rounds from 1-13 and the Crypto spec was numbering the _same_ rounds from 2-15, which more closely matches the numbering in the NIST standard.
Along with the SPIKE code changes, we will also be fixing the proof-of-concept code which was also numbering the rounds from 1-13. The SPIKE code passed the NIST tests because, in this case, two wrongs made a right.

nibrunieAtSi5, 2023-01-07:
> Crypto spec was numbering the same rounds from 2-15

I think the spec is numbering them `2 - 14`.

kdockser, 2023-01-09:
Thanks Nicolas. You are correct, it is 2-14. I have updated my comment above.

kdockser, 2023-01-23:
This has been fixed in the latest SPIKE code: https://github.com/rivosinc/riscv-isa-sim/blob/zvk-vector-crypto/riscv/insns/vaeskf2_vi.h

closed 2023-01-23 by kdockser