riscv/riscv-crypto #284: Software compatibily of vghmac operand ordering

nibrunieAtSi5, 2023-01-26:
OpenSSL implementation:
- https://github.com/openssl/openssl/blob/e44b34185268e99ee2dc499073cf56d0465d60e2/crypto/modes/gcm128.c
```
# define GCM_MUL(ctx)      ctx->funcs.gmult(ctx->Xi.u,ctx->Htable)
# if defined(GHASH_ASM) || !defined(OPENSSL_SMALL_FOOTPRINT)
#  define GHASH(ctx,in,len) ctx->funcs.ghash((ctx)->Xi.u,(ctx)->Htable,in,len)
```
It looks like OpenSSL support either `GCM_MUL` or `GHASH` as basic primitive for gcm128 mode implementation.

Previous discussion https://github.com/riscv/riscv-crypto/issues/192

cmuellner, 2023-01-27:
Your comment describes the situation well.

So, `vghmac.vv` performs the following operation: `Y[i+1] = (Y[i] ⋅ H) ^ X[i+1]`.

OpenSSL requires `clmult()` (multiplication plus reduction) to be implemented.
The `ghash()` call is optional (but it needs to be decided at built time if it is available or not).
If `ghash()` is not provided, then it will be emulated with `gmult()` (as you have mentioned in your comment):

```c
        /* Emulate ghash if needed */
        for (i = 0; i < len; i += 16) {
            memcpy(tmp, &inp[i], sizeof(tmp));
            Xi[0] ^= tmp[0];
            Xi[1] ^= tmp[1];
            funcs.gmult(Xi, Htable);
```

So `ghash()` is a loop (processing N 128-bit blocks) over `gmult()` plus a preceding XOR operation. I.e. `Y[i+1] = ((Y[i] ^ X[i])  ⋅ H)`.

Linux uses the same idea. An example that implements this is the [AArch64 backend](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm64/crypto/ghash-ce-glue.c?h=v6.2-rc5#n71), were we have the following code:
```c
	do {
		const u8 *in = src;

		if (head) {
			in = head;
			blocks++;
			head = NULL;
		} else {
			src += GHASH_BLOCK_SIZE;
		}

		crypto_xor((u8 *)&dst, in, GHASH_BLOCK_SIZE);
		gf128mul_lle(&dst, &key->k);
	} while (--blocks);
```

This raises the question of how `vghmac.vv` can be integrated into such `ghash()` functions.
The [current approach](https://github.com/openssl/openssl/pull/20149/files#diff-424d01288cbe5b43a874172872c7cb15866520fe1c95ca1e4b6bed8782c7dbd8R317) is to peel off the first and the last half loop iterations and do the following:
```
# void gcm_ghash_rv64i_zvkg(u64 Xi[2], const u128 Htable[16],
#                           const u8 *inp, size_t len);
gcm_ghash_rv64i_zvkg:
    vsetivli x0, 4, e32, m1, ta, ma
    vle32.v $vH, $Htable
    vle32.v $vXi, $Xi
    # First loop part
    vle32.v $vinp, $inp
    vxor.vv $vXi, $vXi, $vinp
    add $inp, $inp, 16
    add $len, $len, -16
    beqz $len, Lend
Lstep:
    vle32.v $vinp, $inp
    add $inp, $inp, 16
    add $len, $len, -16
    vghmac.vv $vXi, $vinp, $vH
    bnez $len, Lstep
Lend:
    # Final multiplication (no XOR operation)
    la $TMP0, Lzero
    vle32.v $Vzero, $TMP0
    vghmac.vv $vXi, $Vzero, $vH
    vse32.v $vXi, $Xi
    ret
```

So the first half is a `vxor.vv` and the last half becomes a `vghmac.vv` with a zero-block as XOR operand.

If we would have `vghash.vv`, which does `Y[i+1] = ((Y[i] ^ X[i])  ⋅ H)`, the resulting code would become:

```
    vsetivli x0, 4, e32, m1, ta, ma
    vle32.v $vH, $Htable
    vle32.v $vXi, $Xi
Lstep:
    vle32.v $vinp, $inp
    add $inp, $inp, 16
    add $len, $len, -16
    vghash.vv $vXi, $vinp, $vH
    bnez $len, Lstep
Lend:
    vse32.v $vXi, $Xi
    ret
```

kdockser, 2023-01-27:
Thanks Christoph and Nicolas,

I think the code above is not quite right as the GHASH calculation starts off with a multiplication of vinp[0] and H. The current approach already takes this into account. The proposed approach needs to bypass the first XOR by forcing vxi[0] = 0.

If we increment the input pointers in the starting sequence of the current approach, the main loop is identical in both approaches. The ending code is different for the current approach as we have used up all of the cipher text outputs and so we need to feed in a zero as the last vinp value. 

The current approach
```
# void gcm_ghash_rv64i_zvkg(u64 Xi[2], const u128 Htable[16],
#                           const u8 *inp, size_t len);
gcm_ghash_rv64i_zvkg:
    vsetivli x0, 4, e32, m1, ta, ma
    vle32.v $vH, $Htable
    vle32.v $vXi, $vinp     //vXi[0] = inp[0]
    add $inp, $inp, 16
    add $len, $len, -16
Lstep:
    vle32.v $vinp, $inp
    add $inp, $inp, 16
    add $len, $len, -16
    vghmac.vv $vXi, $vinp, $vH
    bnez $len, Lstep
Lend:
    # Final multiplication (no XOR operation)
    la $TMP0, Lzero
    vle32.v $Vzero, $TMP0
    vghmac.vv $vXi, $Vzero, $vH
    vse32.v $vXi, $Xi
    ret

```

The proposed approach
```
    vsetivli x0, 4, e32, m1, ta, ma
    vle32.v $vH, $Htable
    la $TMP0, Lzero
    vle32.v $vXi, $TMP0 // vXi[0] = 0
Lstep:
    vle32.v $vinp, $inp
    add $inp, $inp, 16
    add $len, $len, -16
    vghash.vv $vXi, $vinp, $vH
    bnez $len, Lstep
Lend:
    vse32.v $vXi, $Xi
    ret

```

nibrunieAtSi5, 2023-01-27:
I think the `vghmac.vv` made sense because I considered fusing the very last operation (see figure) which made the peeling more useful.
[image]
This approach does not seem compatible with most software implementations which rely on an elementary `ghash_update` like approach. As indicated by @kdockser this makes the counter approach epilog longer than the proposed approach.
I think there is a need for two extra `vghash` in the proposed approach, one to hash in the message length and one to hash in the final E_K(0) that I don't see in any of the proposals.

I think the proposed `vghash` (ADD then MUL) makes more sense (ease of software integration, and compatibility with the ghash update scheme) and I don't think it prevents us from implementing a future parallel evaluation scheme (it just modifies the epilog).

I think we may want to consider an extra instruction with xor-with-zero (`gf128mul_le`-like).

By the way why not use `vmv.v.i $vXi, 0` to 0-ify a vector register (group) ?

kdockser, 2023-01-27:
Perhaps I am missing something, but I'm still not seeing how `vghmac.vv`  doesn't fit with existing software. If we can add gcm_ghash_rv64i_zvkg (as defined above) to OpenSSL with `vghmac.vv` as-is, why is there a need to redefine the instruction?

From the examples I have seen, there is no use of a `ghash` function in architecture-independent code that requires the inputs to be `vXi[n]`, `vinp[n]`, `vH` for each 128-bit block of cipher text.  Instead, it seems like architecture-specific code is called upon to perform the _entire_ ghash calculation across all blocks of ciphertext. If that is the case,  it shouldn't matter if the inputs to `vghmac.vv` are  `vXi[n]`, `vinp[n+1]`, `vH`. Is there ever a case in architecture-independent code (that architecture-specific code cannot replace) where a `ghash` function is called upon to process a single block of cipher text?

Please show me where OpenSSL and/or Linux code mandates that the ciphertext input to ghash code must be  from the current iteration (i.e., `vXi[n]`) and not the next  iteration (i.e., `vinp[n+1]`).

cmuellner, 2023-02-01:
> I think the code above is not quite right as the GHASH calculation starts off with a multiplication of vinp[0] and H. The current approach already takes this into account. The proposed approach needs to bypass the first XOR by forcing vxi[0] = 0.

We have it integrated into OpenSSL and Linux. It passes tons of self-tests in both projects. This does not mean, that the code is perfect, but at least a good indicator that the integration is not entirely wrong.

> The proposed approach needs to bypass the first XOR by forcing vxi[0] = 0.

The first multiplication always uses `GCM_MUL()` in OpenSSL (see function `CRYPTO_gcm128_encrypt` in `crypto.modes/gcm128.c`. Note, that the API is written such that data can be processed in one or more chunks. So the assumption that `Xi[0] = inp[0]` does not hold.

The diagram that @nibrunieAtSi5 posted shows exactly what OpenSSL is doing: `Auth data 1` is multiplied by `H` and then `GHASH()` is called to do one or more iterations of the pink-highlighted operations. The `Xi` is the data coming from "left" (i.e. the input to XOR) and `inp` is pointing to `Ciphertext [m..n]`.

> By the way why not use vmv.v.i $vXi, 0 / vmv.v.i $vZero, 0 to 0-ify a vector register (group) ?

Yes, that's a better way to do that. Thanks!

> I'm still not seeing how vghmac.vv doesn't fit with existing software

The issue is not "doesn't fit", but "could be improved to better fit to long existing SW APIs".

> If we can add gcm_ghash_rv64i_zvkg (as defined above) to OpenSSL with vghmac.vv as-is, why is there a need to redefine the instruction?

To improve the compatibility to existing APIs in significantly wide-used OSS projects.

> Please show me where OpenSSL and/or Linux code mandates that the ciphertext input to ghash code must be from the current iteration (i.e., vinp[n]) and not the next iteration (i.e., vinp[n+1]).

The OpenSSL implementation can be found in `CRYPTO_gcm128_encrypt()` (in[ `crypto/modes/gcm128.c`](https://github.com/openssl/openssl/blob/master/crypto/modes/gcm128.c#L725)). Note, that this function does not represent a complete calculation, but is just a function to process a chunk of user data.

In Linux the abstractions are a bit more structured, but it works the same way. In essence, a [`struct shash_alg`](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/include/crypto/hash.h?h=v6.2-rc6#n134) needs to be implemented, which includes the callback `int (*update)(struct shash_desc *desc, const u8 *data, unsigned int len);`. Like in the OpenSSL case, the function can be called multiple times.

cmuellner, 2023-02-01:
My proposed change to Zvkg would be to introduce two instructions:
* `vgmult.vv vd, vs2`...Multiplication in GF(2^128)
* `vghash.vv vd, vs2, vs1`...XOR followed multiplication in GF(2^128)

OpenSSL's `gmult()` and `ghash()` functions could be implemented in the following way:
```
gcm_gmult_rv64i_zvkg:
    vsetivli__x0_4_e32_m1_ta_ma
    vle32_v $VS2, $Htable
    vle32_v $VD, $Xi
    vgmult_vv $VD, $VS2
    vse32_v $VD, $Xi
    ret

gcm_ghash_rv64i_zvkg:
    vsetivli__x0_4_e32_m1_ta_ma
    vle32_v $vH, $Htable
    vle32_v $vXi, $Xi
Lstep:
    vle32_v $vinp, $inp
    add $inp, $inp, 16 
    add $len, $len, -16 
    vghash_vv $vXi, $vinp, $vH
    bnez $len, Lstep
    vse32_v $vXi, $Xi
    ret
```

This code is not a draft, but tested code, that passes all OpenSSL tests.

I have implemented the whole proposal in QEMU, Binutils and OpenSSL.
Encodings are made up (for obvious reasons), but do not conflict with ratified RISC-V instructions.

You can find the corresponding changes in the following repositories:
* QEMU: https://github.com/cmuellner/qemu/tree/riscv-zvk-ghash
* Binutils: https://github.com/cmuellner/binutils-gdb/tree/riscv-zvk-v2-ghash
* OpenSSL: https://github.com/cmuellner/openssl/tree/riscv-zvk-ghash

If you are interested in testing this, I can recommend the following test suite, which is executed in a few seconds and includes many GCM tests:
```
# Assuming Binutils, QEMU and OpenSSL are already built

cd path/to/openssl

# Enable all Zvk* extensions in QEMU
export QEMU_CPU=rv64,zba=true,zbb=true,zbc=true,zbs=true,zbkb=true,zbkc=true,zbkx=true,zkn=true,zks=true,v=true,vlen=256,x-zvkb=true,x-zvkg=true,x-zvknha=true,x-zvknhb=true,x-zvkns=true,x-zvksed=true,x-zvksh=true                          
# Enable all Zvk* extensions in OpenSSL (i.e. tell OpenSSL to execute optimized routines)
export OPENSSL_riscvcap=rv64gcv_zba_zbb_zbc_zbs_zvkb_zvkg_zvknha_zvknhb_zvkns_zvksed_zvksh

# Run the EVP AES test suite, which includes tests for all cipher modes (ECB, CBC, GCM, etc.)
path/to/qemu-riscv64 test/evp_test -iter 1 ./test/recipes/30-test_evp_data/evpciph_aes_common.txt
```

To make sure the Zvkg instructions are indeed executed, you can disable the instructions in QEMU (set `x-zvkg=false` in the `QEMU_CPU` string), keep the `OPENSSL_riscvcap` variable as is, and run the tests again. You will get `SIGILL` errors, which are caused by illegal instructions, which can only be the caused by Zvkg instructions.

ptomsich, 2023-02-01:
Given the various implementation choices for a ghash (i.e. gf-mult prefixed or postfixed with a gf-add), I can only reiterate on our discussion in last week's crypto-vector meeting: splitting the gf-add (i.e. the already existing vxor.vv) into a separate instruction allows to either pre- or postfix the gf-mult with this processing step.

We can give guidance as part of the specification to schedule these instructions back-to-back to make it easier for implementors to fuse them into macro-ops.

kdockser, 2023-02-01:
Thanks to everyone for the input.

The question before us is _not_ whether an alternative form of the instruction can be used in OpenSSL and Linux but whether the currently defined `vghmac.vv` _can_ be efficiently used in them. This issue cannot be resolved until code is written that uses the current instruction as efficiently as possible. 

As for adding a _gfmult_-type instruction, we had considered that at the time that we defined `vghmac.vv` but decided against it as the current `vghmac.vv` instruction already provides this capability when the vs2 holds "0". Creating a zeroed-out register is trivial as it is just `VXOR vd, vt, vt` and only needs to be executed once before any number of  `vghmac.vv` instructions are executed as a _gfmult_-type instruction.

kdockser, 2023-02-02:
OK, I think I have been able to make sense of it all.

First of all, `vghmac.vv` (let's call it MAC) performs `Y[i+1] = (Y[i] ⋅ H) ^ X[i])`, as Christoph originally said. The alternative approach that is being recommended (Let's call it ACCMUL) is `Y[i+1] = ((Y[i]^ X[i]) ⋅ H)`. The difference is not in whether `X[i]` or `X[i+1]` is being  added, it is in the definition of `Y[i]`.

We originally decided on the MAC approach for this instruction since GHASH  is a series of multiply-adds (see the diagram in Nicolas' post above), starting with _Auth Data 1_ and _Ciphertext 1_  as the multiplicand and addend respectively. This would make Y[0]=(_Auth Data 1_) and would require that the ACCMUL-based code to start with X[0]=0 so that the first ACCMUL only performed a multiply.  However, looking at the OpenSSL C code more closely, it appears that Y[0]= (_Auth Data 1_⋅ H) from a multiplication that is performed before gcm_ghash_rv64i_zvkg is called.

With that understanding, along with Christoph's explanation that the `ghash` function doesn't necessarily perform "all rounds" of the NIST defined GHASH, it seems that the C code is defining Y[i] according to ACCMUL semantics.

So, it seems that the best course of action is go with the ACCMUL approach to match the expectations of OpenSSL and Linux. As Philipp mentions above, there is value in having a standalone gfmult-type instruction as this enables implementations to fuse this with an XOR instruction should implementors want the MAC semantics. Alternatively we could offer both ACCMUL and MAC instructions. Let's discuss this further and close on this issue in tomorrow's Crypto TG meeting.

kdockser, 2023-02-04:
In yesterday's Crypto TG meeting we discussed this and decided that we need to have an ACCMUL instruction as that fits in with the needs and requirements of OpenSSL and Linux code for GCM. But, we also didn't want to be bent too much by the choices that were made by a few libraries (even if they enjoy widespread usage). So, we decided to keep the current MAC instruction as well.

_TLDR: skip to the final paragraph_

However, when I went to add the ACCMUL instruction, I found that we had used up all of the encoding space that had been carved out for Vector crypto instructions with three source operands. While we could ask for more space, I noticed that we still have space for instructions with two source operands. So I thought back to our discussion of adding a GHASH-MUL instruction. While this could be emulated with an ACCMUL instruction with the X[i] input = 0, there is still a benefit to having a standalone GHASH-MUL instruction: as Philipp had pointed out, this would enable implementers to fuse this instruction with a subsequent XOR, which would give the functionality of the MAC instruction.

Since the MAC instruction is just an ACCMUL with X[i] tied to zero, it makes sense to use the same inputs as in ACCMUL. This means that vd and vs2 would  point to the two source operands, just like our AES, and vsm4r instructions. This allows us to share the same funct6 encoding and differentiate this instruction by bits in the vs1 field of the encoding. Basically, we can add this instruction at minimal cost as we would use an encoding that would otherwise be unusable, and implementers could reuse the hardware for ACCMUL.

While we could decide to just have the GHASH-MUL instruction which coders could precede or follow with an XOR to emulate an ACCMUL or MAC instruction, this would significantly limit AES-GCM (or SM4-GCM) performance. While implementations could fuse these cases into a single instruction ACCMUL and/or MAC instruction, this seems like a big ask when it is so easy to provide an ACCMUL instruction.

I didn't want to wait until next week's meeting to resolve this so I took the liberty of updating the specification to replace MAC with ACCMUL and add a MUL instruction.   Please respond ASAP if you object to (or approve of) this change.

cmuellner, 2023-02-04:
I guess it is no surprise that I support this change!

nibrunieAtSi5, 2023-02-04:
I am Ok with this change (even If I like the fact that MACs are easier to pipeline and overlap in uarchs), the ease of software integration sold it to me.

nibrunieAtSi5, 2023-02-04:
I would prefer the name `vghash` to `vghsh` (we already have "long" 6-letter mnemonic such as `vclmul`)

kdockser, 2023-02-06:
Originally we had named the `vgmac` instruction `vghash` but the TG decided against this because --- as you pointed out --- it is just one iteration of the loop within the full algorithm.

closed 2023-02-06 by kdockser