riscv/riscv-crypto #290: Question about AES instruction .vs and .vv form.

JerryShih, 2023-02-12:
Why aes round zero only has scalar form(vaesz.vs)?
In aes middle and final round, they both have scalar and vector form(.vs and .vv).
It's not consistent.

```
vaesz.vs

vaesem.[vv,vs]
vaesef.[vv,vs]
vaesdm.[vv,vs]
vaesdf.[vv,vs]
```

Do we need the .vv form in aes?
I think we will use the same key schedule around the whole data encryption/decryption. If we try to run the aes round in parallel(e.g. CTR mode), we still use the same round key for all aes block.

nibrunieAtSi5, 2023-02-12:
> Why aes round zero only has scalar form(vaesz.vs)?

Because the `.vv` form can be implemented by `vxor.vv`

nibrunieAtSi5, 2023-02-12:
`vaesz` is only a vector xor. Wasting some encoding for something which could be implemented by an existing instruction did not make much sense.

nibrunieAtSi5, 2023-02-12:
> Do we need the .vv form in aes?

The task group idea was to enable the use of vector for parallel encryption/decryption of multiple streams with possibly different keys. We did not want to restrict the ISA to accelerating single stream cryptography.

JerryShih, 2023-02-12:
Got it. Thanks you.

nibrunieAtSi5, 2023-02-12:
Others from the task group may have different answers and may bring more elements.
@JerryShih you can leave this question open if you want a broader set of answers (and a review of my feedback).

JerryShih, 2023-02-12:
Another question about `vaesz.vs`.

The crypto related instructions have the following constraint:
```
This instruction must always be implemented such that its execution latency does not depend on
the data being operated upon.
```
If we use `vxor.vv` for `vaesz.vs`, the `vxor.vv` also needs this constraint. The `vxor.vv` in rvv 1.0 spec doesn't have this constraint. I think we need to setup the constraint for `vxor.vv`.

nibrunieAtSi5, 2023-02-13:
That is a good remark, I think the spec should extend this Kt data-independent timing constraint to `vxor.vv`.

nibrunieAtSi5, 2023-02-17:
I think this could also be extended to `vand.vi` (which can be used as the removed `vandn.vi`).
I think the topic was discussed during Feb 16th TG (I could not attend).

nibrunieAtSi5, 2023-02-17:
I think this should be discussed as its own issue and I have open https://github.com/riscv/riscv-crypto/issues/293 to do so.

kdockser, 2023-02-17:
Jerry, thanks very much for reminding us of this general need!
Since Nicolas has opened a new issue to track the more general need for Data-Independent Instruction Latency of non-crypto vector instructions, I am closing this issue.

closed 2023-02-17 by kdockser