riscv/riscv-crypto #293: Requirement for data-independent latency for instructions outside the new extensions

nibrunieAtSi5, 2023-02-17:
It seems a few instructions outside of the ones newly specified may require to expose data-independent latencies:
- `vxor.vv` (as implementation of `vaesz.vv`)
- `vand.vi` (as implementation of `vandn.vi`)
- others to be listed

During Feb 16th Task Group was discussed the possibility of defining `Zvkt` to list vector crypto and standard vector instructions  with such requirement. I am opening this GH issue to track discussions.

kdockser, 2023-02-17:
Thanks Nicolas.

To add some more information to what Nicolas stated above: 
Since the Vector Crypto Specification already has a Data-Independent Execution Latency (DIEL) requirement for all instructions defined therein, we are working on something analogous to `Zkt` but for (non-crypto) Vector Instructions that could be used for the implementation of cryptographic algorithms, including those _not_ addressed specifically by the Vector Crypto Extensions Specification.

kdockser, 2023-02-17:
Adding to the list above:

1. All logical operations
2. Add operations (e.g, as used in SHA2)
3. Permute and merge operations - on the data being moved, not on the permutation "map" of the elements
4. Multiply - especially instructions useful for arbitrary precision arithmetic
5. Shift - on the data, not the shift amount

NB: DIEL does not apply to non-data parameters of the execution such as: vl, vtype, VLEN, ta, ma. We are just preventing the leaking of data (e.g., plaintext, keys, intermediate values) through timing.

allenjbaum, 2023-02-17:
This is probably known and obvious to you, but the exceptions to permute,
merge, and shift ops don't matter only if
 - they are constants in the code, or
 - fixed by the algorithm outside of the data stream (so they could be
register based if not dependent on the data, but rather on the number of a
round or something).

kdockser, 2023-02-17:
Thanks Allen. You are correct, the latency due to the "control" operands for these instructions don't matter when they don't contain that data that we are trying to protect. I was alluding to cases where these operands are constants defined by the algorithm, as is common in cryptography.
My biggest concern was with instructions like `vrgather` where the mapping of the elements can have a big impact on latency. Simpler instructions that stay in their own lane (e.g., shift) could probably just be defined as DIEL for all inputs without imposing an undue burden on implementers - however, it really is the data inputs that we are concerned about.

nibrunieAtSi5, 2023-02-19:
Useless factoid: I think we can already implement a DIEL `vmerge.vvm vd, vs2, vs1, v0.t` for SHA-2:

- with a `vrev8.v vd, vs1, v0.t` by selecting SEW=8b and first copying `vs2` into `vd`: with SEW=8b, the `vrev8` has no effect except selecting active elements from vs1 (since `vmerge` selects active elements from `vs1` and not `vs2`, the latter must be copied into `vd` first).
- with a `vandn.vv vd, vs1, x0, v0.t` by selecting the same SEW as SHA-2 and first copying `vs2` into `vd`: inverting and splatting `x0` will give a all1s pattern transforming the `vandn` into a merge operation based on the mask `v0.t`.

`Zvkt` will allow us not to use this trick and save us a `SEW`/`vl` change.

nibrunieAtSi5, 2023-02-23:
@kdockser and @mjosaarinen are working on specifying `Zvkt` (see https://github.com/riscv/riscv-crypto/blob/master/doc/vector/riscv-crypto-vector-zvkt.adoc).
Closing this issue.

closed 2023-02-23 by nibrunieAtSi5