riscv/riscv-crypto #316: Add Zvbb and Zvbc to RISC-V profiles

nhukc, 2023-04-24:
Zvbb and Zvbc should be required in RVA23. Devices that do crypto and run a Linux kernel will require these instructions.

Tracking this as an issue so it doesn't get forgotten.

Zvbb and Zvbc should be added to https://github.com/riscv/riscv-profiles/
Extension names that have been invalidated also need to be updated in the profile docs.

nibrunieAtSi5, 2023-04-25:
The extension are mandated as part of the options for RVA23: `Zvkn` and `Zvks` although there are not mandatory (nor optional) as standalone extensions. Are you recommending that they be mandatory as `V` is ? 
I think the task group and the committee (If I understood correctly) prefer to have new extensions optional in a profile in a given year before making them mandatory in the next profile version.

nhukc, 2023-04-25:
Oh I missed that they're included in Zvkn. That seems fine then.

closed 2023-04-25 by nhukc

nibrunieAtSi5, 2023-04-25:
link for reference: https://github.com/riscv/riscv-crypto/blob/v20230424/doc/vector/riscv-crypto-vector-zvkn.adoc