riscv/riscv-crypto #332: vghsh vs vgmul

aamartin0000, 2023-06-25:
If I correctly understand the Sail code in the spec, isn't vgmul functionally the same as vghsh, with vs1=0?  If so, would it be worth saving an opcode, instead having a "vx" version of vghsh?  There could be an assembler pseudo op that equates vgmul.vv to "vghsh.vx vd,vs2,x0".

nibrunieAtSi5, 2023-06-25:
I think this was considered. One of the issue is that the `.vx` version would only make sense for `x0`, as for standard implementations (ELEN,XLEN=64) the scalar register would be too small to fit anything useful (except 0). It may be better to keep the encoding with func3 for `.vx` for another operation which can benefit fully from this form.

aamartin0000, 2023-06-26:
Ok, that's fair.  Would it be worth adding a note to the vgmul spec mentioning this?  It might be a useful hint for implementation and verification.

aamartin0000, 2023-06-26:
That should be good enough!  I didn't go back to this spec after I wrote my rtl code.

I'll go ahead and close this issue.

closed 2023-06-26 by aamartin0000