riscv/riscv-crypto #333: is vector masking actually DIEL?

jnk0le, 2023-06-27:
Zvkt spec says:
> However, DIEL does not apply to other values such as vl, vtype, and the mask (as specified by vm).

Does it mean the content of the mask input (held in v0) is exempt from DIEL or is it just the case
of instruction with `vm=1` encoding having different timming then the `vm=0` one?

The instruction list suggests it's the latter option. But in this case Zvkt can have an performance impact in certain non crypto cases (`mu` vs `ma` impact depends on uarch)

nibrunieAtSi5, 2023-06-27:
I think the former "as specified by vm" is intented to mean that `v0` when used as the mask is exempt from DIEL but not when used as a data otherwise covered by DIEL.

Which means that for a masked operation, the actual operation latency may depend on how many bits are set in the mask.

IIRC `v0` cannot be used as both a mask source and an operand source as it would contradict the requirement that a register operand must be used with a single `EEW`.

allenjbaum, 2023-06-27:
I would argue otherwise. If the source of the mask were a fixed, constant
value, I might agree.
But if the mask is created as a result of a compare between vector
elements, and is then used as an operand for a DIEL covered vector op, it's
basically transitively one of the DIEL function operands, so should also
be DIEL - it shouldn't matter how many bits are set,  the calculation
should be data value independent.

nibrunieAtSi5, 2023-06-27:
I think (I might be wrong) than during the task group discussion it was considered that there were no case in the target cryptographic primitives where the mask could be something else than a fixed constant value thus it was not necessary to mandate that it was submitted to the DIEL constant. IIRC we explicitly excluded mask value from the "data value" to allow more implementation freedom.

allenjbaum, 2023-06-28:
I vaguely remember that now - it's been too long. That said - we should be
explicit about that, even if its non-normative text

kdockser, 2023-06-28:
For masked instructions (indicated by the suffix `vm`), latencies can vary due to the effect of the `mask` itself. Put another way, when controlling what is active, the `mask` is not considered a data value for purposes of DIEL. For a given mask, latencies cannot vary for different _data_ values, whether they are in active or inactive elements.
We added the parenthetical "as specified by vm" to clarify that this doesn't apply to masks that are used as source or destination _data_ (e.g., `vmand.vm`).

jnk0le, 2023-06-28:
so, the only "safe" way to consume result of compare&set instructions are
- other mask-ALU instructions
- store to memory
- `vadc`/`vsbc` (considered as data operand?) then classic SSE2 like masking
- cast mask to data by DIEL insn (I think it should be excluded)

kdockser, 2023-06-29:
My typo above (subsequently corrected) may have caused confusion. The example I intended to give above of a "mask" consuming and producing instruction is `vmand.mm`. This instruction only has an `.mm` suffix. 

Instructions with a `.vm` suffix are those that are controlled by a `mask` input. The latency of a `.vm` instruction can vary according to the `mask` input.

There is no special property applied to data (or the register holding the data) that is --- or is intended to be --- used as a mask. Thus, instructions like `vmadd.mm` are still subject to DIEL.

closed 2023-06-29 by kdockser