riscv/riscv-crypto #336: aes_decode_rcon lookup in vaeskf2.vi pseudocode appears to be incorrect.

mlawson-tt, 2023-06-30:
After the info from #267, Spike updated their implementation of `vaeskf2.vi` to handle the 1-based round indexing (so 2-14 instead of 1-13), but in doing so, changed the `aes_decode_rcon` lookup from

`      const uint32_t rcon = kRoundConstants[(round >> 1) - 1];`

to

`      const uint32_t rcon = kRoundConstants[round >> 1];`

but the spec still has `aes_decode_rcon(rnd>>1)`.  

Looking at _A.3 Expansion of a 256-bit Cipher Key_ from the NIST AES spec, the first round constant value used is 0x01, so if the example begins at round 2 (considering there are 13 rounds in the example, and the first round must be even since it performs the round constant lookup), the round constant for round 2 is `0x01`.  However, using the expression from the spec results in a round constant of `0x02` since `rnd>>1` will be 1 for round 2.  Also, the last round constant used is 0x40 in the example, which is consistent with the Spike implementation (if Spike were incorrect, its table would need to be shifted and would then result in values outside `0x01` and `0x02`).

If the reasoning above is correct, it would seem to imply the spec should use `aes_decode_rcon((rnd>>1) - 1)` like Spike does.

nibrunieAtSi5, 2023-07-02:
I think you are right, the pseudo code was not updated when the range was fixed.

https://github.com/rivosinc/riscv-isa-sim/blob/d2cfddd1141ba95adcfc9617e7767513f0f47146/riscv/insns/vaeskf2_vi.h#L74
```
static constexpr uint8_t kRoundConstants[7] = {
  0x01, 0x02, 0x04, 0x08, 0x10, 0x20, 0x40,
};
(...)
    if (is_even_round) {
      const uint32_t rcon = kRoundConstants[(round >> 1) - 1];
      temp = temp ^ rcon;
    }
```

It looks like this was implemented slightly differently in QEMU:
https://github.com/CodethinkLabs/qemu-ct/blob/b5191056c087a033a75232bcc94778ec09024670/target/riscv/vcrypto_helper.c#L484
```
        static const uint32_t rcon[] = {
            0x01000000, 0x02000000, 0x04000000, 0x08000000, 0x10000000,
            0x20000000, 0x40000000, 0x80000000, 0x1B000000, 0x36000000,
        };

        (...)

        if (uimm % 2 == 0) {
                    (...)
                    rcon[(uimm - 1) / 2];

```

With the same behaviour: imm of `2` is mapped to table entry 0, immediate of `4` is mapped to entry `1`, ...

nibrunieAtSi5, 2023-07-02:
I filed https://github.com/riscv/riscv-crypto/pull/337 to align the pseudo-code such that the pseudo-code `aes_decode_rcon` helper function can be used in `vaeskf2.vi` pseudo-code to be conformant with AES specification.

kdockser, 2023-07-03:
Thanks for finding this Matt.
And, thanks for fixing it Nicholas. 
Accepted pull request.

closed 2023-07-03 by kdockser