riscv/riscv-crypto #339: rs1 mentionned as control operand for vslide1

nibrunieAtSi5, 2023-07-06:
The `Zvkt` specification mentioned that the scalar operands for `vslide1*` is exempt from DIEL:

https://github.com/riscv/riscv-crypto/blob/master/doc/vector/riscv-crypto-vector-zvkt.adoc

> The values in rs1 and uimm are used for control (i.e., slide amount) and are exempt from DIEL.
> 
> vslide[up|down].v[ix]
> 
> vslide1[up|down].vx

However in the case of `vslide1` this operand is not really part of the control, as it not the slide amount but the scalar value being inserted, should we clarify that this remark only applies for `vslide[up|down].v[ix]` ? (the spec has the same presentation for `vrgather*` where the remark applies to both items and `vslide*` where it should apply to only the first item)

kdockser, 2023-07-06:
Thanks Nicolas.

You are correct. Since the scalar is data and not control, it is not exempt for DIEL. This applies to the `vslide1[up|down].vx` instructions and the similar `vfslide1[up|down].vf` instructions. I will update the spec accordingly.

kdockser, 2023-08-03:
This has been updated in the spec.

closed 2023-08-03 by kdockser