riscv/riscv-crypto #342: Is "Vector Bit-manipulation used in Cryptography" an accurate name for Zvbb?

topperc, 2023-07-17:
As I understand it Zvbb was originally Zvkb and late some non-cryptography instructions like vcpop.v, vclz.v, and vctz.v were added to it to form Zvbb.

I think the cryptography reference should be dropped from the expanded name and instead it should be something like "Vector Basic Bit-manipulation".

nibrunieAtSi5, 2023-07-17:
I agree that it would make more sense to remove the mention to "cryptography", in particular because the extension is no longer prefixed by `Zvk`. However I do not know if this is just a cosmetic change or if it will have larger impact on other deliverables required in the on-going ratification process (which could make the change less straightforward).

nibrunieAtSi5, 2023-07-17:
If we go done this road we may need to update the description of the extension as well (which should have less impact):

> General purpose bit-manipulation instructions which are essential for implementing common cryptographic workloads securely & efficiently.

To maybe:
> General purpose basic bit-manipulation instructions, some of which are essential for implementing common cryptographic workloads securely & efficiently and others extend the generic bit manipulation capabilities of RISC-V on vectors. 

https://github.com/riscv/riscv-crypto/blob/master/doc/vector/riscv-crypto-vector-zvbb.adoc

aamartin0000, 2023-07-18:
It might also be worth noting that most of the instructions in Zvbb have corresponding scalar instructions in Zbb (and, in some cases, Zbkb).  It's not a perfect correspondence though.

nibrunieAtSi5, 2023-08-01:
Note: Following a discussion between the Architecture Review Committee and the TG chair, a new sub-extension containing the instructions for the original `Zvkb` is going to be split out of `Zvbb`. This may be a good time to clarify naming.

kdockser, 2023-08-03:
Thanks for the suggestion, Craig. We have renamed Zvbb as you suggested.

The specification has been updated such that Zvkb is a proper subset of Zvbb. Zvkb is the mnemonic for "Vector Cryptography Bit-manipulation" and Zvbb is the mnemonic for "Vector Basic Bit-manipulation"

closed 2023-08-03 by kdockser