riscv/riscv-crypto #344: Zvbc: Define behaviour of `vclmul[h].[vv,vx]` in RV32 mode

cmuellner, 2023-07-17:
What are `vclmul[h].[vv,vx]` instructions, that are part of Zvbc, supposed to do in RV32 mode?
The spec states that these instructions are reserved encoding if `SEW is any value other than 64`, which does not necessarily rule out RV32.

This has been raised on the Binutils mailing list by Jan Beulich: https://sourceware.org/pipermail/binutils/2023-July/128414.html

nibrunieAtSi5, 2023-07-17:
I could not find any mention of `XLEN` nor `ELEN` in the spec (I thought we discussed the case RV32 at some point but could only find this RFC https://github.com/riscv/riscv-crypto/blob/master/doc/supp/vector-clmul-rfc.md), this might indeed require some clarification.

The issue might lie with the `.vx` variants. For integer operations, the vector crypto spec states:
> For integer operations, the scalar can be taken from the scalar x register specified by rs1. If XLEN>SEW, the least-significant SEW bits of the x register are used, unless otherwise specified. If XLEN<SEW, the value from the x register is sign-extended to SEW bits.

https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#10-vector-arithmetic-instruction-formats

In the `vclmul[h].v[v,x]` case it would make more sense to zero-extend (to be confirmed) when `XLEN=32`.

kdockser, 2023-07-24:
Thanks for the question, Christoph. The answer lies in the pseudo code (Which I will reformat in the spec to make more readable.)
```
let op1 : bits (64) = if suffix =="vv" then get_velem(vs1,i)
                      else zext_or_truncate_to_sew(X(vs1));```

The xv form of the instruction reads X(vs2) and zero-extends it to SEW=64. When you need something other than 0s in the upper 32 bits of the vs1 operand, the vx form will not be very useful in RV32. However, the vv form will continue to operate.

closed 2023-07-24 by kdockser