riscv/riscv-crypto #357: Are any of Zvkn/Zvks/Zvkg included in Zvkt?

aamartin0000, 2023-08-04:
By omission in the Zvkt spec, can I assume that all of the above subsets are exempt from DIEL?
The scalar crypto Zkn and Zks subsets are explicitly included in the Zkt spec.

nibrunieAtSi5, 2023-08-04:
The spec mentions

> All cryptography-specific instructions defined in this Vector Crypto specification (i.e., those in [zvkned], Zvknh[ab], [Zvkg], [Zvksed] and [zvksh] but not [zvbb],[zvkb], or [zvbc]) shall be executed with data-independent execution latency as defined in the RISC-V Scalar Cryptography Extensions specification.

This may need to be repeated in `Zvkt`

nibrunieAtSi5, 2023-08-04:
Reference https://github.com/riscv/riscv-crypto/blob/master/doc/vector/riscv-crypto-spec-vector.adoc#1-extensions-overview

aamartin0000, 2023-08-04:
Ok, thanks.  I just looked specifically at the Zvkt adoc, so I missed that.

closed 2023-08-04 by aamartin0000

nibrunieAtSi5, 2023-08-04:
> Ok, thanks. I just looked specifically at the Zvkt adoc, so I missed that.

No worries.