riscv/riscv-crypto #358: Question about vaeskf1 pseudo code

Mingku-Chang, 2023-08-08:
Hi,
I am studying Vector Crypto Specification 1.0.0-rc1.
I am confused about the pseudo-code of vaeskf1.
The variable "CurentRoundKey" appears to have 4 elements with a width of 8 bits.
The input data of the function "aes_subword_fwd" is 32-bit width.
So, how can I treat the statement "aes_subword_fwd(CurrentRoundKey[3])"?

nibrunieAtSi5, 2023-08-08:
`CurrentRoundKey` is a 4-element array (`[3:0]`), each being 32-bit wide.
It appears the pseudo-code notation is not consistent with other pseudo code in the spec (e.g. `vsha2c` uses
```
  let MessageShedPlusC[3:0] : bits(4*SEW) = get_velem(vs1, 4*SEW, i);
```
where `<array>[3:0] : bits(4*SEW)` is used to split `4*SEW` bits into a 4-element array, each element being SEW bits wide.

To be consistent `bits(32)` should be replaced by `bits(128)` (Element Group Size for the round key)

kdockser, 2023-08-08:
Fixed in vaeskf1 and vaeskf2.

closed 2023-08-08 by kdockser

Mingku-Chang, 2023-08-09:
Hi @nibrunieAtSi5 & @kdockser,
Thanks for your help.
I guess the "foreach" statement in vsesfk1 is unnecessary.

nibrunieAtSi5, 2023-08-10:
@Mingku-Chang , the `foreach` statement in https://github.com/riscv/riscv-crypto/blob/master/doc/vector/insns/vaeskf1.adoc is still required as it iterates over the various active element groups (not the word in a group). For example if `vl=12` (assuming for example VLEN=256, LMUL=2, SEW=32-bit), then `vaeskf1.vi` is still going to perform 3 parallel key schedule expansion and produce 3x128-bit.