riscv/riscv-crypto #364: Question about zvknha and zvknhb extension.

JerryShih, 2023-09-20:
If one platform support SHA-512(zvknhb), will it also imply having zvknha?
The zvknhb could support both SHA-512 and SHA-256. Will the qemu or linux kernel need to enable SHA256 with zvknha extension or just use single zvknhb extension setting?

e.g.
```
# Will the following command have sha256?
qemu-riscv64 -cpu rv64,g=true,f=true,d=true,v=true,vlen=128,elen=64,vext_spec=v1.0,x-zvknhb=true
# Should we have the explicit sha256 setting?
qemu-riscv64 -cpu rv64,g=true,f=true,d=true,v=true,vlen=128,elen=64,vext_spec=v1.0,x-zvknha=true,x-zvknhb=true
```

nibrunieAtSi5, 2023-09-20:
> Do we have sha256 with single "ZVKNHB" extension?
Yes

>  Should we have both "ZVKNHA" and "ZVKNHB" cpu setting in qemu if we try to have both sha256 and sha512?
No, those extensions are exclusive and cannot be enabled at the same time

nibrunieAtSi5, 2023-09-20:
From https://github.com/riscv/riscv-crypto/blob/main/doc/vector/riscv-crypto-vector-zvknh.adoc

> Zvknhb supports SHA-256 and SHA-512.

JerryShih, 2023-09-20:
Got it, thanks.

closed 2023-09-20 by JerryShih

nibrunieAtSi5, 2026-02-02:
⚠️ the answer to this question may actually change if https://github.com/riscv/riscv-isa-manual/pull/2635 make it through the review process.