riscv/riscv-crypto #365: Question about CLMUL/CLMULH pseudo code

Mingku-Chang, 2023-09-28:
Hi,

I found a slight wrong with the pseudo-code of the clmul/clmulh.
vs1/rs1 is the multiplier, and vs2 is the multiplicand.
Their positions on the pseudo-code are reversed compared to ISA-sim.
```
  ISA-sim                  pseudo-code
  --------------------     --------------------
  for (i=0~63)             for (i=0~63)
    if (vs1[i])              if (vs2[i])
      vd ^= (vs2 << i)         vd ^= (vs1 << i)
```

nibrunieAtSi5, 2023-10-03:
Hi @Mingku-Chang , thank you for reporting this.

Honestly, I would think the order between multiplicand and multiplier is arbitrary and ISA sim and the pseudo code using a different order (since the multiplication is commutative) will not make any difference.

kdockser, 2023-10-05:
I agree with Nicolas, the answer is the same either way so there is no need to change the pseudo-code.

closed 2023-10-05 by kdockser