riscv/riscv-crypto #375: Is there a benchmark for vector crypto extension?

lhtin, 2023-11-06:
The current repository has a scalar version of the benchmark. Is there a vector version of the benchmark in the community or a plan for it? If there is any, we are willing to participate, and thank you very much.

grnewell, 2023-11-06:
> EXTERNAL EMAIL: Do not click links or open attachments unless you know the content is safe
>
> I remember SiFive doing this test

There are a very wide range of implementation points for vector crypto, with small to wide datapaths and pipelining options.  I don’t know if any implementers have published any benchmarks on their proprietary designs.  From an instruction count PoV it provides a very substantial savings, but I don’t know where that is recorded, either.  Perhaps someone else in the task group can find those.

G. Richard Newell
Assoc. Technical Fellow, FPGA Business Unit, Microchip Technology
(408) 643-6146 (office), (408) 882-4785 (mobile), +1 (925) 478-7258 (Skype, not monitored)
PGP: (2009 DSA-1024, ELG-4096) B751 FC13 8B4E 49DA 2270 35A2 20E4 E66A D0D0 2E34
     (2016 SSA-4096, RSA-4096) 65F5 CCD6 23B3 BD3D CEDE AB58 171F F4DE E7D0 3ECA

aamartin0000, 2024-01-25:
OpenSSL has implementations for both RISC-V scalar and vector instructions.  Those could be used for "apples to apples" comparisons between implementations.  Note that, due to implementation-specific decisions (e.g. instruction latencies, vector width) this code may not be optimal for any particular implementation, requiring tuning.