riscv/riscv-crypto #378: test suites for vector crypto?

aamartin0000, 2024-01-11:
Hello, is there anywhere, particularly in the RISC-V github, that has tests for the vector crypto instructions?

I'm interested in both correctness (e.g., make sure that every entry in the sbox generators is correct) and performance.  I realize that the performance testing has sensitivity to the implementation (instruction latencies, throughput, etc.) but would like to see a starting point.

nibrunieAtSi5, 2024-01-11:
Hi Al,
   Have you tried the code samples https://github.com/riscv/riscv-crypto/tree/main/doc/vector/code-samples ? They are self checking.

aamartin0000, 2024-01-11:
Thanks, Nicolas.  I missed that.  I will forward this to my verification and performance colleagues.

nibrunieAtSi5, 2024-01-11:
You are welcome @aamartin0000.
Note that this is not the vector crypto Architecture Compatibility Test (ACT) which is not available yet.