riscv/riscv-crypto #380: Clarify description of Zvkt

Mingku-Chang, 2024-01-24:
I am confused with the Zvkt (Vector Data-Independent Execution Latency) definition.
It said, "All floating-point operations" are not affected by Zvkt.
But the vfslide1up.vf and vfslide1down.vf operations are listed in the affected range.
I think these 2 operations are in floating-point format, even though they were categorized as Vector Slide Instructions.
For the same reason, I believe vfmv* (Floating-Point Scalar Move) operations are also unaffected by Zvkt.
Could you clarify this part?

nibrunieAtSi5, 2024-01-25:
@kdockser for your consideration. (we could/should clarify since `vfmv.v.f` is indeed listed in RVV 1.0 Section 13 but `vfslide1*` are listed in Section 16).

mjosaarinen, 2024-01-25:
My recollection is that the use case for the slide instructions in vector crypto was to help with AES key scheduling (to hold subkeys) -- and this is obviously unrelated to floating point. So it seems that the inclusion of this instruction is indeed an error.

nibrunieAtSi5, 2024-02-02:
To be honest, it could go either way for me. I found the use of floating-point registers as extra storage a bit of an anti-pattern but I am quite sure it will happen. 

I found your argument about not making any normative change if we can avoid it spot on. The spec being ratified, the bar to make such change should be pretty high and I do not think this qualify.

So overall this sounds very reasonable to me.

We could maintain a list of known _inconsistencies_ that we may eventually revise in a future revision of the vector crypto spec. And in general gathering community feedback on usage would be interesting.