riscv/riscv-crypto #381: Illegal condition for VWSLL operation

Mingku-Chang, 2024-01-24:
I guess that the VWSLL operation with (SEW == 64) is not allowed.
Vector Crypto follows a lot of regulations from Vector.
Therefore, results can't exceed 64-bit width.
I recommend adding some statements to show readers that Vector has such a regulation.

nibrunieAtSi5, 2024-01-24:
You are right `vwsll` with SEW=64-bit (`vsew=3`) would mean an effective element width of 128-bit (EEW=2*SEW) which is not currently supported so this behaviour is reserved.

Note that they are other possible restrictions, for example if ELEN=32 then SEW=32-bit is also reserved (Since `Zvbb` could be supported alongside `Zve32x` for example). In that case I am not sure a specific mention of `SEW==64` really make sense since it is not the only possible limitation. We could (should ?) mention that this instructions inherits element width constraint from RVV.

nibrunieAtSi5, 2024-02-22:
@kdockser for your consideration

wmat, 2024-03-28:
Is this recommendation still valid?  If yes, please transfer issue to riscv-isa-manual repo or let me know to do it.  Thanks.