riscv/riscv-crypto #384: Integration of Scalar into the Unpriv spec has began.

wmat, 2024-02-15:
Note that I have began integrating the Scalar spec into the Unprivileged spec.  I'll update this issue with a Pull Request reference against riscv-isa-manual when I have one.

nibrunieAtSi5, 2024-02-15:
Thank you @wmat

mjosaarinen, 2024-02-16:
Thanks, added a note in README.md as discussed in the TG meeting. https://github.com/riscv/riscv-crypto/commit/ef1eeb9074e57d4e08d77d3db43dc345e354bc7f

ben-marshall, 2024-02-16:
Great to hear @wmat :+1:

ThinkOpenly, 2024-03-06:
Process question: given there are human-readable documentation and Sail descriptions for both scalar and vector, is there a specified order as to how things get integrated into the canonical repositories? Does vector depend on scalar? Does the documentation have to be updated before the Sail descriptions are added, or are they done simultaneously?

(My real question is when to expect the Sail specifications to be updated, but want to learn the process as well. Thanks!)

wmat, 2024-03-06:
Let me try to answer as I understand the current specifications.  As I understand it, scalar & vector are separate specifications and do not necessarily reoly on each other.  Any SAIL code in either spec would be self contained and have been manually added.  At present SAIL descriptions would need to be manually updated in the applicable scalar crypto or vector crypto chapter if they change in the SAIL model.

latifbhatti, 2024-03-07:
I hope this message finds you well. I wanted to provide some insight into the use of Sail pseudo code rather than actual Sail code in the RISC-V Vector Cryptography Extension specification.

As stated in the Vector Crypto spec, the original intention was to include actual Sail code in the specification. However, due to the absence of an approved RISC-V Vector Sail Model at the time of completing the Vector Cryptography extension specification, there was no basis on which to build the required Vector Sail model. (section of riscv vector crypto spec 1.0 (1.2. Sail Specifications)). its mean riscv vector extension is important to implement vector crypto spec 1.0 extension.

only zvkt extension of vector crypto extension depended on scalar crypto extension.
reference: (Chapter 2. Extensions Overview (riscv vector crypto(1.0)))  "It is important to note that the Vector Crypto instructions are independent of the implementation of the Zkt( (including Zvbb, Zvkb, and Zvbc) extension and do not require that Zkt is implemented" and detail discussion in 2.15. Zvkt - Vector Data-Independent Execution Latency section.

I share this information based on my understanding. If you have any concerns or questions about this information, please feel free to discuss them with me. I appreciate your attention to this matter. If you have additional information to share, I would be grateful for your input.

Thank you for the conversation.

ThinkOpenly, 2024-03-08:
> [...] scalar & vector are separate specifications and do not necessarily reoly on each other. Any SAIL code in either spec would be self contained and have been manually added. At present SAIL [...]

I intended to distinguish between the human-readable ISA spec, and the Sail reference _itself_ (https://github.com/riscv/sail-riscv). It sounds like you are referring just to the ISA spec, am I correct?

When should one expect the Sail code here to be integrated into that RISC-V Sail reference repository?

wmat, 2024-03-08:
Bill Traynor
Documentation Build and Release Engineer
RISC-V International

> [...] scalar & vector are separate specifications and do not necessarily
> reoly on each other. Any SAIL code in either spec would be self contained
> and have been manually added. At present SAIL [...]

Yes, the ISA spec only.

When should one expect the Sail code here to be integrated into that RISC-V
> Sail reference repository?
I’m hoping to have the option to pull in SAIL code late this year.

wmat, 2024-03-12:
Quick question, where Bitmanip instructions are included in the Scalar Crypto spec, should they be included when integrated as a chapter?  The reason I ask is that the Bitmanip spec will be a chapter of it's own shortly and this will create duplication I think.

wmat, 2024-03-13:
> Quick question, where Bitmanip instructions are included in the Scalar Crypto spec, should they be included when integrated as a chapter? The reason I ask is that the Bitmanip spec will be a chapter of it's own shortly and this will create duplication I think.

NVM, there's no duplication.  I see that the Bitmanip content is referenced directly from the Bitmanip repository.  This will need to be updated to reference the Bitmanip chapter once it's integrated.

wmat, 2024-03-14:
I'm ready to merge the scalar-crypto chapter but I'd like to get someone to briefly skim it to make sure nothing jumps out as wrong or incomplete.  Note that the content hasn't changed at all, I've simply refactored the asciidoc to fit into the higher level Unprivileged manual.

You can grab the PDF here: https://github.com/riscv/riscv-isa-manual/actions/runs/8268811976/artifacts/1323222855

If that PDF has already been deleted, let me know and I'll generate another for you.  I'd appreciate a LGTM response here before I merge it.

Thanks

wmat, 2024-03-15:
Note that I'll be merging scalar-crypto on Monday, March 18th.

wmat, 2024-03-28:
Scalar crypto is now integrated into the Unprivileged spec.  Note that I'll begin asking for Issue closure or transfer to the riscv-isa-manual repo when an issue is still valid.  Following that, this repository will be archived.

ThinkOpenly, 2024-03-28:
@wmat where would one expect to find the Sail implementation for this?

wmat, 2024-03-28:
@ThinkOpenly I asked Bill McSpadden and he say in the [SAIL repo](https://github.com/riscv/sail-riscv), the files of interest are:
riscv_types_kext.sail
riscv_insts_z*.sail

ThinkOpenly, 2024-03-28:
> @ThinkOpenly I asked Bill McSpadden and he say in the [SAIL repo]

Thank you for that information!

However, I should've been more specific. I'm looking for the Sail implementation of Zvkned, for example.

(If I am asking in the wrong context, perhaps you could redirect me.)

wmat, 2024-03-28:
There would appear to be an open [PR for that implementation](https://github.com/riscv/sail-riscv/pull/234).

wmat, 2024-09-06:
> The descriptions for the "zip" and "unzip" instructions have been swapped
> in the larger Unpriv spec, compared to the latest Crypto-Scalar spec.
>
> The instruction encodings are also different.
>
> (Credit for finding this goes to PColim
> <https://www.reddit.com/user/PColim> on Reddit)

Yes, I saw that on reddit as well.  Will fix that up shortly.