riscv/riscv-crypto #387: Incorrect information in Vector Crypto Specification Release 1.0.0

michaelmaitland, 2024-04-24:
I downloaded the PDF from the [Vector Crypto Specification Release 1.0.0](https://github.com/riscv/riscv-crypto/releases/tag/v1.0.0) and noticed it had the typo that had been fixed in https://github.com/riscv/riscv-crypto/commit/ae384d56a4e70d8f5279777ce9898a697ab1a0c3.

Is there the ability to fix this in the generated PDF? I understand there's been a big effort recently to merge all ratified extensions into riscv-isa-manual (where it is fixed), but I think it should still be fixed here, especially since that code change has been committed.

nibrunieAtSi5, 2024-04-24:
This repo has been archived and I am not sure we are going to generate a new spec from it. It might be clearer to remove the release version from here and points to the main repo. @wmat, @kdockser ?

wmat, 2024-04-25:
I'm not exactly sure the best way to do this.  I could add a link to riscv-isa-manual in the README.  But are you asking me to remove the last 1.0.0 release?

michaelmaitland, 2024-04-25:
> I could add a link to riscv-isa-manual in the README

Given that the 1.0.0 release is not the same as what is checked in the repo and that we don't plan to fix 1.0.0 release or put out a new one in this repo, I wouldn't be against removing the last release.

That being said, if the README said that the 1.0.0 release is not up to date with the code in the repo and to use riscv-isa-manual instead, that would have been enough to make me go read the riscv-isa-manual which does not have the error.