riscv/riscv-crypto #390: vl and vstart constraints on vaes*/vsha*/vgh*/vsm* instructions

ved-rivos, 2024-06-05:
> Instructions that violate the vl or vstart requirements are reserved.

What is the behavior(s) specified/allowed if these instructions are issued when this state happens to be  _reserved_?

ved-rivos, 2024-06-08:
@kdockser

nibrunieAtSi5, 2024-06-08:
My understanding is that the behavior is undefined in this case: e.g. it is not mandatory to raise an exception in that case.
If I remember a discussion with @aswaterman correctly (@ mention to correct me if I am wrong), this is not even affected by `Sstrict` as this does not fall into the unimplemented opcode category.

ved-rivos, 2024-06-08:
Not requiring an exception will be preferred. Raising exception based on VL leads to implementation challenges.

nibrunieAtSi5, 2024-06-08:
I agree, if I recall this is what let to the reserved rather than mandating an exception during the task group discussion.

nibrunieAtSi5, 2024-06-08:
From a security point of view, some "definition" of undefined behavior might be  problematic.

ved-rivos, 2024-06-08:
Would the code invoking such an instruction not be already flawed if it is invoking say AES with a message that is a multiple of the block size? A correct implementation would have padded the message using techniques such as ciphertext stealing or other appropriate methods depending on the protocol.

nibrunieAtSi5, 2024-06-08:
I believe you are right @ved-rivos and any vl corruption would have bad consequences way before we get to these element group instructions.

aswaterman, 2024-06-09:
Reserved means unspecified, not undefined. A given implementation still needs to have some defined behavior (trap, round down vl, deterministically execute some custom instruction, etc.).

I agree with Ved that this does not exacerbate the security concern since the software was broken anyway.

allenjbaum, 2024-06-09:
I'd actually like to understand why trapping is an implementation
challenge.
That doesn't necessarily mean it is in a state that can be recovered from,
does it?
And, it would be a fatal trap - but at least you would know the SW is
broken.

ved-rivos, 2024-06-09:
> Reserved means unspecified, not undefined. A given implementation still needs to have some defined behavior (trap, round down vl, deterministically execute some custom instruction, etc.).

Agreed. Rounding down the vl or executing as custom instruction such as a no-op could be some defined behaviors.

> I'd actually like to understand why trapping is an implementation challenge.

Trapping based on vl would require an implementation to have ability to cause redirects from all vector pipes that have these ALUs. For SW it would be quite evident it is broken as the output for invoking with a sub-block would not be sensible - e.g., encrypt or decrypt will fail.

closed 2024-06-21 by ved-rivos